hlorus/CAD_Sketcher #228: [BUG] The Sketcher data failed when I copy a Sketch to another file

Byvm815, 2022-07-19:
### Contact Details

2529025686@qq.ccom

### Description

I 'm no very good at English, with Chinese :
当我把一个Sketcher绘制的物体拷贝到另一个文件时，它的数据丢失了。

### Addon Version

0.24.0

### Blender Version

3.1.2

### What platform are you running on?

Mac OSX

Byvm815, 2022-07-20:
I read the roadmap, it will be true on release 2.0.0

exegetor, 2022-07-22:
google translate of the Chinese:
"When I copy an object drawn by Sketcher to another file, its data is lost."

Perhaps you copied only the mesh?  If so, the mesh will not carry the Sketcher data with it.

hlorus, 2022-07-22:
Ahh thanks for translating. Sketcher data is currently stored on the scene object so you would have to copy the whole scene to another file if you want to reuse the sketcher data.

Byvm815, 2022-07-23:
@exegetor 
You are right. I know what happened in the next day. And I'm sorry to   you(@hlorus ) for my slowness.
你说得对，我在第二天就明白怎么回事了,抱歉这么晚才回复你。

Byvm815, 2022-07-23:
> Ahh thanks for translating. Sketcher data is currently stored on the scene object so you would have to copy the whole scene to another file if you want to reuse the sketcher data.

@hlorus 
Thank you very much for your help
非常感谢你的帮助

exegetor, 2022-07-23:
@hlorus, can you drop the bug label and relabel it Documentation?   I think this ought to be in the manual for 1.0

hlorus, 2022-07-23:
> @hlorus, can you drop the bug label and relabel it Documentation?   I think this ought to be in the manual for 1.0

I agree, however are you not able to edit the issue? I should look into github permissions then.

exegetor, 2022-07-23:
That is correct; I could find no means of relabeling this.  Thanks!

closed 2023-11-15 by hlorus