hlorus/CAD_Sketcher #242: [BUG]: unable to create sketch

michael-barrow, 2022-08-06:
**Issue:**
Crash after adding new sketch. Cannot edit a sketch.

**Version:**
Blender: 3.2.1 or 3.0.1. with CAD Sketcher 0.24.0, Ubuntu 22.04 (Fresh install)

**Steps to reproduce**
ctl+shift+a, select any option or tool from within blender, sketch disappears, with no ability to add any elements to sketch.

**Example**

[image]

exegetor, 2022-08-06:
when you select the workplane, it has to be highlighted (white).  It's not clear that this is happening here.  If you click without a highlighted outline, it does not register the click (interprets as a cancel)

michael-barrow, 2022-08-07:
Thanks for the response @exegetor. It seems it is not possible for me to select the workspace, it simply vanishes as in the example .gif when moused over (highlighted white) and left clicked.

amrsoll, 2022-08-09:
I agree with exegetor. As far as I can see from the video you are moving your mouse around to fast to be selecting a workplane.

It might help you to open the side panel to have a view of the active sketches.

[image]

You have to hover the selector closer to the black borders of the gray squares, then the gray square (workplane) should be highlighted in white.

Closing for now, unless you continue struggling with this, feel free to reopen.

closed 2022-08-09 by amrsoll