hlorus/CAD_Sketcher #243: [BUG] Cad-Sketcher dependency install process might install to user site-packages instead of blender - missing PYTHONUSERSITE in install scripts.

laundmo, 2022-08-09:
### Contact Details

laundmo#7544 on discord

### Description

When installing the py-slvs package from within the addon, the wrong pip will be called.

Explanation of why this might happen here:
https://github.com/robertguetzkow/blender-python-examples/tree/master/add_ons/install_dependencies#installing-the-package

The simple fix to this is to modify the environment variables with which pip commands are run.

This blender console script shows the issue:

```py
>>> import sys
>>> sys.executable
'C:\\Program Files\\Blender Foundation\\Blender 3.2\\3.2\\python\\bin\\python.EXE'

>>> import subprocess
>>> subprocess.check_output([sys.executable, "-m", "pip", "-V"])
b'pip 22.2 from C:\\Users\\laundmo\\AppData\\Roaming\\Python\\Python310\\site-packages\\pip (python 3.10)\r\r\n'
```
as you can see, this found the wrong pip (from my appdata site-packages)

heres a minimal version of the fix in action:

```py
>>> import os
>>> environ_copy = dict(os.environ)
>>> environ_copy["PYTHONNOUSERSITE"] = "1"
>>> subprocess.check_output([sys.executable, "-m", "pip", "-V"], env=environ_copy)
b'pip 21.2.4 from C:\\Program Files\\Blender Foundation\\Blender 3.2\\3.2\\python\\lib\\site-packages\\pip (python 3.10)\r\r\n'
```

### Addon Version

latest

### Blender Version

all

### What platform are you running on?

Windows

laundmo, 2022-08-09:
I'm open to making a PR for this myself over the next weekend - i'd just like a go-ahead first.

amrsoll, 2022-08-09:
oh wow, good catch, that could also explain some "breakages" for people with multiple blender versions and so on...

laundmo, 2022-08-09:
Aye, stumbled across this on the Discord server where a user was having issues with the isntall proceess using *condas pip* which doesn't work at all outside of a conda environment. Its potentially huge to the install process breaking.

hlorus, 2022-08-09:
If you could submit a PR that would be great obviously.

hlorus, 2024-07-20:
Thanks for the report. With blender 4.20 CAD Sketcher is installed as an extension which brings support for pip dependencies. Blender will install the correct solver module when the extension is installed.

closed 2024-07-20 by hlorus