hlorus/CAD_Sketcher #244: Macintosh and Linux is separate not metge

mac4kent, 2022-08-09:
### Contact Details

N/A

### Description

A bug happened!Look at picture they are wrong!
[image]

Mac/Linux will not merge and not work.
They did give us link. https://hlorus.github.io/CAD_Sketcher/advanced/#console-output
Blender.app is NOT INSTALL. Only way drag app into Application folder. 
See picture. 
[image]

Linux/Windows is merge yes install Blender.exe only.
Mac must separate from Linux and separate from Windows.

### Addon Version

N/A

### Blender Version

3.2.2

### What platform are you running on?

Mac OSX

amrsoll, 2022-08-10:
Would you like to write a paragraf to improve the Mac instructions to install?

mac4kent, 2022-08-10:
What is improve , instructions ?
I am not good english text only I am full deaf and low vision.

mac4kent, 2022-08-10:
@amrsoll

KeithSloan, 2022-09-04:
Only vaguely related but I tried to install CAD-Sketcher and am failing at the first hurdle(s).

Trying to follow the instructions at https://hlorus.github.io/CAD_Sketcher/installation/

Okay I click on the "Zip Archive" but being on a Mac this automatically gets expanded in $HOME/Downloads.
Okay I use compress to create a zip file and then I think I am able to install the Addon.

I assume it is not working because I do not have blender with solverspace installed.
( In the past I have installed solverspace as an app I assume from a dmg but it was some time ago )

The instructions say
It's recommended to use the addon with Blender installed from the official website.
What does that mean? I have to install a copy of Blender with solverspace already installed or just
that blender needs to have been installed from official website?

Then it gives the option Install from PyPi - Press. "Install from PyPI" but I see nothing like that to press.
Try manually installing with pip install py-slvs. Restart Blender but no solverspace in Bender Preferences.

I am obviously too dumb for modern software.

hlorus, 2023-01-23:
The installation guide already has a note "do not unpack it after downloading". Should that be expanded to say that it might have to be re-zipped?

edgimar, 2023-02-22:
> The installation guide already has a note "do not unpack it after downloading". Should that be expanded to say that it might have to be re-zipped?

Yes, that would make sense to include.  Also, for me using the install from PyPI button worked, but after it was installed, I needed to restart Blender for it to be recognized.

hlorus, 2023-02-22:
I hope this is clear enough: https://github.com/hlorus/CAD_Sketcher/pull/340

> Also, for me using the install from PyPI button worked, but after it was installed, I needed to restart Blender for it to be recognized.
There should already be a massage when this operator finishes which says to restart blender

closed 2023-11-15 by hlorus