hlorus/CAD_Sketcher #245: [BUG] Enable 0.25 not possible

abergmeier, 2022-08-10:
### Contact Details

abergmeier@gmx.net

### Description

Added 0.25 from zip but cannot be activated.

Seems to die in `icon_manager.py:73`.

### Addon Version

0.25.0

### Blender Version

3.2.1

### What platform are you running on?

Linux

hlorus, 2022-08-10:
Could you post a screenshot of the console?

ra100, 2022-08-20:
Seems that the "icons" folder can't be found, I've copied all folders from `ressources` one level higher, and addon could be enabled without the error.

hlorus, 2022-08-21:
The zip seems to be correct, at least theres a folder with icons in that directory. Maybe try completely removing the addon folder and adding it back from the zip. There could be some strange stuff going on when it's just overwritten e.g. old files laying around.

ra100, 2022-08-21:
I'm not sure what's wrong, but when I download the zip from here: https://github.com/hlorus/CAD-Sketcher/archive/refs/heads/main.zip, I get this content:

[image]

and icons_manager has the proper link `filepath = Path(Path(__file__).parent, "ressources/icons")` in there

I've been trying to remove addon and then install back, didn't work. But now, I've tried to remove addon from older config version of blender, downloaded zip again, and seems working.
It looks weird why it didn't work before. Anyway, my issue is solved.

calebmeyer, 2022-11-04:
@ra100's fix worked for me as well. Clicking the checkbox to enable the addon was giving this error:
[image]

And copying icons up fixes it.

Cheaterman, 2023-02-24:
This was most certainly fixed by #337 - please re-open in case you can reproduce it with current CAD Sketcher. As @ra100 noted, it's probably a good idea to entirely remove your CAD Sketcher addon folder before reinstalling.

closed 2023-02-24 by Cheaterman