hlorus/CAD_Sketcher #248: Introduce formatting guidline

hlorus, 2022-08-11:
Automatic formatting can go along way to improve readability of the code and make our lives easier. Following are the steps to get there.

- [ ] Evaluate and decide on options
- [ ] Batch format the project 
- [ ] Add guidline to documentation 
- [ ] Add github action to check formatting on PRs

hlorus, 2022-08-11:
Personally i would go with black as a formatter to keep it simple and avoid trouble with different configurations between contributors.

laundmo, 2022-08-11:
seconding the Black idea. Black has the great advantage of being opinionated and consistent - theres only one parameter to adjust which is max line length and i would leave that at default.

This helps tremendously for preventing inconsistencies across contributors and PRs.

I also recommend using https://pre-commit.com/ as part of the setup to reduce the amount of github action checks failing on formatting or other linters.

hlorus, 2022-08-12:
Okay, let's go with black for now.

hlorus, 2022-08-17:
All done.

closed 2022-08-17 by hlorus