hlorus/CAD_Sketcher #250: Mac on Command, Option (ALT), Control, Shift August 13, 2022

mac4kent, 2022-08-13:
I notice no one body answer about FEEDBACK so I put this here.

Feature this CAD Sketch on Blender 3.2.2 for Mac. Do not use Windows.
I see you made Keymap code on CAD Sketch and I am show you screenshot on Mac.
https://imgur.com/9FiQte1

Your code is https://imgur.com/n0Y6xSh

https://support.apple.com/en-us/HT201236

hlorus, 2022-08-14:
Not exactly what this is about. Afaik mac has an alt key.

mac4kent, 2022-08-14:
Allright
Macintosh        |    Windows
Command       =       CTRL
Option (ALT)   =        ALT
Control            =          ... ? ? ?

amrsoll, 2022-08-22:
In your Apple support link, it does say that Ctrl = Control on Mac

Control = CTRL

[image]

mac4kent, 2022-08-22:
Yes 
Mac is Control (CTRL) = Windows is ???
Mac is Command = Windows is CTRL.

amrsoll, 2022-08-22:
No, Mac Control = Windows Ctrl

[image]

mac4kent, 2022-08-22:
I ask you do you have your own Macintosh computer?

mac4kent, 2022-08-22:
You are wrong answer.
Mac Command + C is copy
Mac Control + C is nothing 
Mac Command + V is paste.
Mac Control + V is nothing.
Windows CTRL + C is Copy
Windows WindowKey + C is nothing
Windows CTRL + V is Paste
Windows WindowKey + V is nothing.
Mac is not same as Windows.

mac4kent, 2022-08-22:
https://support.apple.com/en-us/HT201236#:~:text=Command%2DC%3A%20Copy%20the%20selected,for%20files%20in%20the%20Finder.

mac4kent, 2022-08-22:
[image]

amrsoll, 2022-08-23:
What is your problem then? Are you not able to use the shortcuts?

mac4kent, 2022-08-24:
You did not answer. I am ask you question "Do you have your own Macintosh computer?"

closed 2023-01-23 by hlorus

mac4kent, 2023-01-24:
I see you hate Mac! Because you not want work on Mac keyboard. You do love Windows only.

hlorus, 2023-01-24:
This doesn't have anything to do with the developers opinion of  mac. Your bug report doesn't clearly describe the problem. It's not clear if this is just about the documentation or if a shortcut is broken. Also the links to the images no longer work.

mac4kent, 2023-01-24:
Screencast Key Add-Ons already fix a bug 4 months ago for Blender 3.4.1 support for Mac.

mac4kent, 2023-01-24:
Yes.