hlorus/CAD_Sketcher #255: Add copy/paste operators

hlorus, 2022-08-19:
Allows to copy selected entities and their constraints and copy them at a later point.
This also has to work after the original entities are deleted.

therealjacobbrauer, 2022-09-01:
> Allows to copy selected entities and their constraints and paste them at a later point.
> This also has to work after the original entities are deleted.

Does this mean that you are going to be copying selected sketch entities and their constraints and pasting them, and then deleting the original ones without losing the copies, 

**-or-**

 just copying and then deleting the original ones, and then some time down the road pasting the saved copies that have not been visible the rest of the time, since the originals were deleted?

hlorus, 2022-09-01:
> > Allows to copy selected entities and their constraints and paste them at a later point.
> > This also has to work after the original entities are deleted.
> 
> Does this mean that you are going to be copying selected sketch entities and their constraints and pasting them, and then deleting the original ones without losing the copies, (this is what solid works supports)
> 

Yes, this one. It would simply take a snapshot when copying. After that there's no connection to the original entities.

therealjacobbrauer, 2022-09-01:
Excellent. Would a option for a start/end point of the copy operation be okay to proceed with as well? 

also is an option desired for the copy operation so that there is a way to do copy or move entities, like in a text editor this would be referred to as copy or (cut = move) entities?

hlorus, 2022-09-01:
> Excellent. Would a option for a start/end point of the copy operation be okay to proceed with as well? 
> 
> also is an option desired for the copy operation so that there is a way to do copy or move entities, like in a text editor this would be referred to as copy or (cut = move) entities?

Such things would be great but i advise to keep it simple to start with and then add additional features once a basic version has been reviewed/merged.

closed 2022-11-19 by hlorus