hlorus/CAD_Sketcher #256: Add operator to align a workplane to the 3d cursor

hlorus, 2022-08-19:
Aligns selected workplane to the 3d cursor. Maybe it should somehow be possible to set the cursor while being in the solvespace select tool.

therealjacobbrauer, 2022-09-01:
Okay so I'm going to spell out a path to accomplish this issue, please @hlorus let me know if this path is correct.

hlorus, 2022-09-01:
Are you suggesting that the user copies the values from the panel and paste it somewhere?

IMO that would be a rather slow workflow. The idea here is to create a operator which does this in one step.

hlorus, 2022-09-01:
Note that CS currently doesn't have a concept of an active entity. Therefore i'd suggest to display this operator in the context menu of workplanes.

therealjacobbrauer, 2022-09-01:
> Are you suggesting that the user copies the values from the panel and paste it somewhere?
> 
> IMO that would be a rather slow workflow. The idea here is to create a operator which does this in one step.

No I wasn't suggesting that the user copies the values from the panel and paste it somewhere, it was more of pseudocode to  have a developer write a function that grabs the values when the user selects the operator "align workplane with 3d cursor". My apologies about not being clear on my reply.

therealjacobbrauer, 2022-09-01:
> Note that CS currently doesn't have a concept of an active entity. Therefore i'd suggest to display this operator in the context menu of workplanes.

what in my spelled out path to accomplish finishing this issue prompted you to think that I was asking for BCS (blender_cad sketcher) to have a concept of an active entity? I am not familiar with the term active entity and I am willing to edit it out of my proposed solution I just don't understand which part of my proposed solution you are specifically referring to. 

thanks for getting back to my replys so quickly!!

hlorus, 2022-09-01:
> what in my spelled out path to accomplish finishing this issue prompted you to think that I was asking for BCS (blender_cad sketcher) to have a concept of an active entity? I am not familiar with [...]

That was just a general note. Blender has the concept of an active object. As that's not available in BCS the operator will need to have a property to specify which workplan it should act on. "slvs_entity_pointer" can be used to achieve that.

Regarding your described approach: Not sure what's ment by "shift+D pad". Also incrementally rotating workplanes seems a bit out of the scope here.

>did you mean to say align just the angle of the selected workplane to the 3d cursor, or also the position of the workplane to the 3d cursor...?

I'd say both but would be great to have a setting to just apply the location or rotation.

commit 76b582a6 (2022-11-04): Ops: Add align workplane to cursor operator

Closes #256
closed 2022-11-04 by hlorus