hlorus/CAD_Sketcher #257: Offset Tool

hlorus, 2022-08-19:
Add a tool to offset selected entities. 
- Geometric constraints should be copied
- Dimensional constraints should be skipped
- Arcs that get to small should get merged into a corner

amrsoll, 2022-09-11:
As a novice CAD user, could you please elaborate on what this tool should do?

As I understood from Discord, the offset tool duplicates entities + adds a distance constraint between the original entities and the new entity group.

Does that mean we need to figure out entity groups and constraints between them??

hlorus, 2022-09-12:
> Does that mean we need to figure out entity groups and constraints between them??

The offset tool should simply duplicate a set of entities together with the constraints they're holding and offset them by the given value. There are no groups involved tho, only regular entities.

To keep it simple we could just apply the operation on all selected entities, later it would probably be nice to automatically find all connected segments.

amrsoll, 2022-09-12:
Ok, that seems pretty simple. Do we expect to set the offset with a mouse drag or specific input values?

Relates to #273 #255 , but without having a copy-buffer and ctrl-C / ctrl-V behaviour

amrsoll, 2022-09-13:
> and offset them

This means we are just translating all the entities by the same vector, right? Just making sure because I am hearing different stories

hlorus, 2022-09-13:
> > and offset them
> 
> Or is it supposed to behave like [on this video](https://youtu.be/BJQ9f_WqDWc?t=59)? (which to me looks more like an outset)

Exactly, maybe the wording is a bit misleading...

JesseRigon, 2022-11-13:
The wording is not misleading, it's misunderstood. In the engineering and CAD/CAM fields an offset can be either in or out, positive or negative. It simply refers to "off" the base line

bonjorno7, 2023-11-13:
Not sure if this warrents its own issue so I'm putting it here:
[image]
When using the offset tool on an arc like this it gives an error.

hlorus, 2023-11-16:
Thanks @bonjorno7, i've pushed a fix! Btw there is a WIP PR to improve the offset tool #400, i could use some help there if you'd like to take a look :)

ShnitzelKiller, 2024-02-14:
It seems that currently the offset tool does not copy constraints, even though the tooltip says it does. For example, bevel one corner of a rectangle and then use the offset tool:

[image]

Not even the horizontal and vertical constraints are copied.
[image]