hlorus/CAD_Sketcher #258: Reference Dimensions

hlorus, 2022-08-19:
### Discussed in https://github.com/hlorus/CAD_Sketcher/discussions/84

<div type='discussions-op-text'>

<sup>Originally posted by **TheMakerTales** April 30, 2022</sup>
A "construction" mode for dimensional constraints. Sometimes you want to know how long or round a part of a sketch is but not actually contain it.</div>

amrsoll, 2022-08-24:
> contain it

Did you mean "constrain it" ?

exegetor, 2022-08-24:
What if we have a checkbox "fixed", like we do with entities?
If it's unchecked (free-floating), ignore it during mouse drags, then recalculate the value on mouse release?

I'd also recommend a new theme color for the 3d view, and a color on the panel display also to differentiate fixed/floating measurements

amrsoll, 2022-08-24:
I'm being a bit dim today :zany_face: I hadn't read "Reference" keyword.

I'd like to know from hlorus if he agrees that a new colour is the best way forward.

amrsoll, 2022-08-25:
> What if we have a checkbox "fixed", like we do with entities?
> If it's unchecked (free-floating), ignore it during mouse drags, then recalculate the value on mouse release?

I like that :)

hlorus, 2022-08-29:
I would say use "reference" as the term instead of "construction" which is IMO a bit misleading.
I'm fine with a new theme color. I would additionally also display the constraint value in brackets/quotation marks or something to make it obvious.

therealjacobbrauer, 2022-09-01:
In solid works, they use "evaluate > measure" to look at a value of a dimension in the non editing mode of viewing that value. This is similar to what is being discussed here as "reference > dimension". Personally I feel like having a section, (possibly done as a python class to continue with the cad sketcher coding style), that is dedicated to evaluating sketch entities and solid objects makes sense for down the road in developing CAD_sketcher where more advanced evaluation things can be added in such as obtaining the mass and volume of a 3d object.

amrsoll, 2022-09-03:
@therealjacobbrauer On a semantical point, I would agree with you, but I believe that the constraints objects are close enough (API-wise) that we should reuse them for this purpose. The advanced features are too far down the road to consider at the moment (mass, volume etc..)

See #272 for implementation

PR #272 (2022-09-20): Create "constructor" constraints.
Fixes #258

[image]
[image]

## TODO

- [x] Update constraint value from the attached entity instead of the solver .. hmmm
- [x] Change look
- [x] Make dimension read only when `is_reference` is True.
- [x] Possibly create `class DimensionalConstraint` in `base_constraint.py`
- [x] Fix bug where color is not proper during mouse hover right after switching reference mode
- [x] Revert change where reference constraints are filtered out of the constraints menu
closed 2022-09-20 by hlorus