hlorus/CAD_Sketcher #259: Invoke constraint popup on constraint creation

hlorus, 2022-08-19:
Allows to quickly enter a dimension value when adding a dimensional constraint.
Ideally allows to directly enter a number.

amrsoll, 2022-08-24:
A question about this; I can see a smal dialog that shows on the bottom of the screen when creating a constraint.

Which part of the code is responsible for creating it? Would that window qualify for punching in the numbers if we can give focus to it?

[image]

hlorus, 2022-08-29:
Afaik that's called the operator redo panel. An operator can define a draw method which will be used to draw that panel. There's likeley no support to give it focus, therefor i don't think that's the right approach.

amrsoll, 2022-09-03:
So we we want a pop up right next to the entity/constraint we create in the 3D view?

Example from fusion 360
[image]

hlorus, 2022-09-04:
Yep, well basically just invoke the menu that appears when you click on a constraint.

amrsoll, 2022-09-15:
Founds the proper thing we want to use from the documentation: Panels (and not menus)

https://docs.blender.org/api/current/bpy.types.Panel.html

Which you can spawn using [`Operator.draw()`](https://docs.blender.org/api/current/bpy.types.Operator.html#custom-drawing)

I'll try making a prototype this weekend if nobody else is tackling this :)

hlorus, 2022-09-15:
I've tried implementing the functionality already, spawning the menu isn't a big deal, the bigger problem is to activate the value property so users could directly enter a number. IMO that's needed otherwise it just feels annoying.

hlorus, 2022-09-16:
I've posted a bug report at blender regarding the activate_init issue:
https://developer.blender.org/T101113

amrsoll, 2022-09-16:
I guess we could do a basic text parser for now if only text input work with `activate_init`, might also be helpful for people who want to put units in the input, but that would be secondary .

hlorus, 2022-09-16:
> I guess we could do a basic text parser for now if only text input work with `activate_init`,

True but i wouldn't recommend such a workaround, this task is really just a quality of life improvement.

> might also be helpful for people who want to put units in the input, but that would be secondary .

That's already supported with Integer/Float properties.