hlorus/CAD_Sketcher #261: Angel Constraint Bug - $50

hlorus, 2022-08-19:


amrsoll, 2022-08-22:
Uncaught exception happens when solving the system

[image]

Pointing to [this particular line](https://github.com/realthunder/solvespace/blob/a2b33c81a82cb158db2fa8748bbd161816b23abc/src/entity.cpp#L528)

amrsoll, 2022-08-23:
Even more minimal example using 2 lines, vertical or horizontal constraint + vertical / horizontal Distance constraint

[image]

hlorus, 2022-08-23:
> Compared to the upstream solvesystem repo, the one we use with the python bindings is 1100 commits behind. I think the realthunder/solvespace repo desperatly needs an update from upstream.
> 

Absolutely agree with that, we should probably reach out to realthunder.

> It might be worth trying this version instead: https://github.com/KmolYuan/solvespace
>  https://pypi.org/project/python-solvespace/

That's actually what i used to start with but i've decided to switch to the version that is used now back because it didn't seem to be maintained iirc.

> But I think @hlorus knows best about the different python ports of solvespace.

I don't really know alot about the differences. There's just no available python package of the official solvespace solver as far as i know.

amrsoll, 2022-08-23:
> it didn't seem to be maintained iirc.

The [https://github.com/KmolYuan/python-solvespace](https://github.com/KmolYuan/python-solvespace ) is archived, but he also has https://github.com/KmolYuan/solvespace which is maintained and a lot closer to upstream. I'm trying to use it atm, but there are quite a few differences (obvi)

therealjacobbrauer, 2022-09-01:
> Uncaught exception happens when solving the system
> 
> [image]
> 
> Pointing to [this particular line](https://github.com/realthunder/solvespace/blob/a2b33c81a82cb158db2fa8748bbd161816b23abc/src/entity.cpp#L528)
> 
> Compared to the upstream solvesystem repo, the one we use with the python bindings is 1100 commits behind. I think the realthunder/solvespace repo desperatly needs an update from upstream.
> 
> It might be worth trying this version instead: https://github.com/KmolYuan/solvespace
> 
> https://pypi.org/project/python-solvespace/
> 
> But I think @hlorus knows best about the different python ports of solvespace.

in your code shown here, how are you running the solver.py script? Do you run the solver.py from your python IDE? Sorry for the basic question, I really appreciate your input on these GitHub pages amrsoll.

vespakoen, 2023-01-14:
I git bisected this issue because it seemed to work at some point in the past.
This bug (at least with the first example (https://github.com/hlorus/CAD_Sketcher/issues/261#issuecomment-1222798814, angle_break.blend) seems to be broken by this commit:

https://github.com/hlorus/CAD_Sketcher/commit/4caa0fcd87c4eaf7475adb90a8e0116c4a2106d9#diff-116ba881a2b88b2046ed43e0ccf5660900385cdfd8bb3beea1a41a7cc6c49f83