hlorus/CAD_Sketcher #262: [BUG]

paynterf, 2022-08-20:
### Contact Details

paynterf@gmail.com

### Description

With the new 0.25 update, I tried changing the 'mesh curvature' value in a new sketch.  The initial change (from 12 to 50 in my case) works, but subsequent edits to change the curvature do not succeed.  When I try to us the sketch update button (the curved arrows), I get a popup with a bunch of python errors noted.
[image]
[image]
[image]

### Addon Version

0.25

### Blender Version

3.2.1

### What platform are you running on?

Windows

paynterf, 2022-08-20:
This may have been a user problem.  At some point I discovered I had selected a view that was orthogonal to the workplane (NUM3 instead of NUM7).  Once I figured this out, things started working again.  No idea though why I was able to draw a circle and a line that looked normal, even though I was drawing it orthogonal to the workplane!

closed 2022-08-20 by paynterf

hlorus, 2022-08-20:
This might also have been caused because the object was in editmode. As the object data will be written once you leave editmode other changes might get overwritten.