hlorus/CAD_Sketcher #264: Adding circle on Windows cause Blender to freeze [BUG]

jackjameshoward, 2022-08-22:
### Contact Details

jackjameshoward@skycom

### Description

Adding a single circle allows me to select the centre point and then select the radius.
If a second circle is added after this, the centre can be selected, and then the blender window freezes before selecting the radius.
To clarify, Blender does not report as crashed or not responding, the window looks like it has just stopped refreshing/rendering.
I know Blender is opertating in the back ground because I have had a remote debugging instance running on the CAD Sketcher package and can see that the `View3D_OT_slvs_add_circle2d` `fini()` completes successfully, and I can continue to click and create circles but the Blender windows is frozen so I can't see any of this working. 

I have added the Blender and CAD Sketcher versions below, however, I had the same issue on a previous version of both items.  

### Addon Version

0.25.0

### Blender Version

3.2.2

### What platform are you running on?

Windows

jackjameshoward, 2022-08-22:
[Screen Recording (2).webm](https://user-images.githubusercontent.com/4273996/185909657-3a3c6c10-5724-40f0-be42-bd27e85d521d.webm)

amrsoll, 2022-08-22:
Interesting. Unable to reproduce on Linux (X11)

jackjameshoward, 2022-08-23:
@amrsoll  Unfortunately not, the whole Blender window freezes and I can't view or click anything. 

If you know where I should point a breakpoint in the package code then I could check it in debug.

jackjameshoward, 2022-08-23:
I have noticed if clicked to add a second circle I can see the prompt is to `Radius: Set circle radius`, _but_, if I do not move the mouse then `right-click` to escape command then the window does not freeze. I only happen went the radius tries to set with a mouse move.

amrsoll, 2022-08-23:
Can you confirm (eg through task manager) that it definitely is your gpu and not cpu that is bottlenecking the system when the bug occurs?

jackjameshoward, 2022-08-23:
It doesn't seem to put any load on the GPU. Only the CPU but not a huge amount.

When the system does get bottlenecked the graphics for other applications (browser, vscode) is badly affected. 

https://youtu.be/y0fB2cinScE

amrsoll, 2022-08-23:
oh wow we managed to break the renderer on windows :joy: I would understand if we were hogging up all the resources, but seems bonkers. Could you double check you have the latest drivers for your graphics card? I believe CAD sketcher uses the OpenGL graphics stack

fhanrath, 2022-10-02:
I do have the same (or at least similar) problem on Linux + Wayland (sway/wlroots).
As soon as I try to add a circle (or any other object from the cad_sketcher plugin) into one plane, blender crashes.

I'm using the newest version from the [aur](https://aur.archlinux.org/packages/blender-plugin-cad-sketcher-git) (0.25.0)

The stout is:

```
/usr/include/c++/12.2.0/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slvs_Constraint; _Alloc = std::allocator<Slvs_Constraint>; reference = Slvs_Constraint&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    171135 IOT instruction (core dumped)  blender
```

amrsoll, 2022-10-04:
@fhanrath  I think I am getting confused by the PKGBUILD, could you please try [downloading zip](https://github.com/hlorus/CAD_Sketcher/archive/refs/heads/main.zip) and install it manually?

I am running the exact same graphical stack as you with no issues, so I do not believe that is the issue.

Could you also give us the blender version you are running the addon with?

```
blender --version
```

I think might warrant a separate issue though.

fhanrath, 2022-10-04:
Thank you for the help!

I tried installing the zip file instead of the aur package and it fixed my issue. My bad for not trying this sooner!

I'm not sure where the issue originated from exactly. As far as I can tell, the PKGBUILD copies the contents of this repository to `/usr/share/blender/$_version/scripts/addons/cad-sketcher` (with $_version being `3.3` in my case) and installs the `python-py-slvs` package. The issue may also lie in this package versus the one I installed via pip for the zip.

I don't think anymore that it's an issue with CAD_Sketcher.

If someone has the same issue in the future, here is my blender --version (it's the arch [blender package](https://archlinux.org/packages/?name=blender))
```
❯ blender --version
Blender 3.3.0
	build date: 2022-09-29
	build time: 13:05:52
	build commit date: 2022-09-06
	build commit time: 15:39
	build hash: 0759f671ce1f
	build platform: Linux
	build type: Release
	build c flags:  -Wall -Werror=implicit-function-declaration -Werror=return-type -Werror=vla -Wstrict-prototypes -Wmissing-prototypes -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wlogical-op -Wundef -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Wformat-signedness -Wrestrict -Wnonnull -Wabsolute-value -Wuninitialized -Wredundant-decls -Wshadow -Wno-error=unused-but-set-variable -Wimplicit-fallthrough=5 -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -flto=auto -fopenmp -std=gnu11 -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -ffp-contract=off -msse2
	build c++ flags:  -Wuninitialized -Wredundant-decls -Wall -Wno-invalid-offsetof -Wno-sign-compare -Wlogical-op -Winit-self -Wmissing-include-dirs -Wno-div-by-zero -Wtype-limits -Werror=return-type -Wno-char-subscripts -Wno-unknown-pragmas -Wpointer-arith -Wunused-parameter -Wwrite-strings -Wundef -Wformat-signedness -Wrestrict -Wno-suggest-override -Wuninitialized -Wundef -Wmissing-declarations -Wimplicit-fallthrough=5 -I /usr/include/python3.10 -fopenmp -msse -pipe -fPIC -funsigned-char -fno-strict-aliasing -ffp-contract=off -msse2
	build link flags:  -Wl,--version-script='/build/blender/src/blender/source/creator/symbols_unix.map'
	build system: CMake
```

hlorus, 2023-02-13:
This seems to be solved, thanks alot for the help everyone!

closed 2023-02-13 by hlorus