hlorus/CAD_Sketcher #270: [BUG] trim tool fails with line double-intersecting circle

exegetor, 2022-08-24:
### Contact Details

exegetor@gmail.com

### Description

I was checking Alex's bevel tool fix and I found a new failure mode in the trim tool

Traceback (most recent call last):
  File "C:\Users\shann\AppData\Roaming\Blender Foundation\Blender\3.1\scripts\addons\CAD_Sketcher-main\draw_handler.py", line 79, in draw_cb
    update_elements(context, force=force)
  File "C:\Users\shann\AppData\Roaming\Blender Foundation\Blender\3.1\scripts\addons\CAD_Sketcher-main\draw_handler.py", line 56, in update_elements
    e.update()
  File "C:\Users\shann\AppData\Roaming\Blender Foundation\Blender\3.1\scripts\addons\CAD_Sketcher-main\class_defines.py", line 988, in update
    p1, p2 = self.p1.location, self.p2.location
AttributeError: 'NoneType' object has no attribute 'location'

https://user-images.githubusercontent.com/10360161/186296589-8cf54185-cbf6-43a2-b9ee-9aea55684ba3.mp4

### Addon Version

0.25

### Blender Version

3.1.2

### What platform are you running on?

Windows

TheMakerTales, 2022-08-24:
For sure something funny going on there. For me, it works the first time but if I do an undo and try again it fails like your video above.

amrsoll, 2022-08-26:
It bears saying that the entities still exist, but they lack the location info to draw them :)

[image]

And here is the file to test:

[trim_tool.blend.zip](https://github.com/hlorus/CAD_Sketcher/files/9433954/trim_tool.blend.zip)

hlorus, 2023-01-17:
Afaik this problem is resolved, feel free to reopen if it's still there.

closed 2023-01-17 by hlorus