hlorus/CAD_Sketcher #271: [BUG]

TrabacchinLuigi, 2022-08-24:
### Contact Details

trabacchin.luigi@gmail.com

### Description

i've been unable to draw anything in my file, i've also updated to the new available version i keep getting this error

`Python: Traceback (most recent call last):
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 474, in modal
    return self.evaluate_state(context, event, event_triggered)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 556, in evaluate_state
    return self._end(context, succeede)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\stateful_operator\logic.py", line 617, in _end
    self.fini(context, succeede)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\operators\add_rectangle.py", line 93, in fini
    solve_system(context, sketch=self.sketch)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 271, in solve_system
    return solver.solve()
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 194, in solve
    self._init_slvs_data()
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\solver.py", line 124, in _init_slvs_data
    indices = c.create_slvs_data(self.solvesys, group=group)
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\class_defines.py", line 2400, in create_slvs_data
    return make_coincident(
  File "C:\Users\luigi.trabacchin\AppData\Roaming\Blender Foundation\Blender\3.2\scripts\addons\CAD_Sketcher-main\class_defines.py", line 2354, in make_coincident
    handle = e2.py_data
AttributeError: 'NoneType' object has no attribute 'py_data'
`

### Addon Version

latest

### Blender Version

3.2

### What platform are you running on?

Windows

exegetor, 2022-08-24:
When things go sideways in solver.solve(), it all depends on the geometry in the file.  Can you make a video, drop the .blend file (zip it up 1st), or describe in detail how to replicate the error?

TrabacchinLuigi, 2022-08-24:
it's quite a "complicate" file but it's here on github, so i'll [link it](https://github.com/TrabacchinLuigi/BHT-6000PhysicalAdapter/blob/main/Project.blend). Sorry for not zipping it...

just draw anything in existing or new sketch, it immediately goes boom.

exegetor, 2022-08-24:
Starting with your unmodified file I was able to add, remove, and move elements to the bolt and the long screw without triggering the error.  (blender 3.1.2, CS 0.25)
One of the bolt corners is made coincident to the inner bolt hole instead of the outer perimeter.  I'm sure you want it on the outer perimeter.  See if deleting and replacing this constraint helps.
However, when i tried to delete it, some odd things happened.  Other constraints seemed to be effected as well.
I did see some weird things going on when trying to do this... deleting this constraint made *other* constraints disappear.  I am looking at this and it may be an unrelated bug so I might be making another bug report later.
But see if you can get that constraint correct, and see if that helps the situation.

The file did take a long time to load.  Perhaps with so much going on in the file, it's better to make a blend file for each component, and a main blend file with links to put it all together.  That will keep the sketch data and modifier stack out of the main file (only final meshes will be imported).

TrabacchinLuigi, 2022-08-25:
seems like the bolt (which was the last thing i've drawed) was the problem, i removed some duplicate points and all the constraints and re-constrained in a better way and now everything works. not sure if it's the case to close this thread or should keep it open so you guys can figure out how to handle certain constrain errors by the user, also i'm gonne re-upload the corrected file, so if you need to grab a copy for debug (maybe strip everything else), be sure to get the right version.

amrsoll, 2022-09-15:
Link to the buggy file
https://github.com/TrabacchinLuigi/BHT-6000PhysicalAdapter/blob/012f8f5d33700cc8822b299876ebfb701216d5db/Project.blend

I was able to reproduce the error when trying to move the inner circle of the bolt.

[image]

TimStrauven, 2024-04-25:
Just some extra info on this one, it might be useful:

With this image as a reference:
[image]

The Coincident constraint annotated in red is the issue here. A coincident should have 3 dependencies, in this case that probably should have been [point, circle, sketch]. 
Looking at its dependendencies in this drawing however it is only attached to:
[bpy.data.scenes['Scene'].sketcher.entities.points2D[14], bpy.data.scenes['Scene'].sketcher.entities.sketches[1]]
So only the point and the sketch, somehow the link to the circle was lost.

I think it is not a real solution, but changing the following at line 126 in "solver.py" makes the drawing workable again and gives a bit more insight into it:
```Python
            try:
                indices = c.py_data(self.solvesys, group=group)
                print(f"good constraint: {c} with dependencies: {c.dependencies()}")
            except:
                print(f"deleting faulty constraint: {c} with dependencies: {c.dependencies()}")
                import bpy
                i = context.scene.sketcher.constraints.get_index(c)
                bpy.ops.view3d.slvs_delete_constraint(type=c.type, index=i)
```

After adding the code above and dragging on the drawing it corrects itself and the drawing looks like this:
[image]

The faulty constraint was deleted, but also note how the yellow constraint swapped over (it was always constrained to that circle but it only shows it correctly now)

I think the real issue to solve here is figuring out how the red coincident constraint lost the reference to the circle in the first place. But I am also lost for an answer to that.

hlorus, 2024-04-29:
> I think the real issue to solve here is figuring out how the red coincident constraint lost the reference to the circle in the first place. But I am also lost for an answer to that.

Absolutley! Unfortunatley it's quite difficult to track that down.

I like the idea to add some form of validation of blend files. Doing it automatically in the solver might be problematic however as that would likley add a performance impact. Also it could be confusing to the user if we remove stuff implicitly.
How about a dedicated validate operator that the user can invoke?