hlorus/CAD_Sketcher #274: [BUG]  Redundant constraints result in "Unknown Failure".

ftc, 2022-08-29:
### Contact Details

shawn.meier@gmail.com

### Description

I made a simple sketch with a redundant constraint and noticed that the solver gives me the error message "Unknown Failure: Cannot solve sketch due to unknown failure".

Expected result: A message saying that there are redundant constraints that highlights the constraints that may be removed.  Additionally, I would expect the failure to be more visually noticeable (e.g. colored text or other kind of pop up notification).

Actual Result: "Unknown failure" is displayed in the notification area and the sketch becomes unresponsive (no dragging or other changes seem possible).

[Constraint_failure.blend.zip](https://github.com/hlorus/CAD_Sketcher/files/9446183/Constraint_failure.blend.zip)

### Addon Version

0.25.0

### Blender Version

3.2.2

### What platform are you running on?

Mac OSX

ftc, 2022-08-29:
Note: deleted contact info since it was my email address and removed it from the version history.  Responding to this thread is the best way to contact me for additional info.  I didn't realize the contact info would be posted publicly.

ftc, 2022-08-29:
Note2: based on discussion on the discord, it appears that the conflicting constraints are more red than the normal ones.  I think this is a reasonable way to notify the user, but perhaps some more obvious colors could be used?

With conflicting constraints:

[image]

Without conflicting constraints:

[image].

amrsoll, 2022-09-03:
This is (un)fortunately the responsibility of the solver we are using. We are not mapping all the correct error codes properly atm, this is probably fixed with #246

See #246 

>         "Constraints seem redundand"

amrsoll, 2022-09-08:
Forgot to replay to your question @ftc  :) 

> but perhaps some more obvious colors could be used?

The colors are configurable in the Add on settings ;)

amrsoll, 2022-09-08:
Aaaand I misunderstood the original issue!

redundant constraint != Conflicting constraints

I feel like this doesn't necessarily requires addressing before the 1.0 release; As explained, there are highlight colors and warning icons for the concerned constraints :smile:

This request might be more interesting when tackling first-time user experience in more depth

amrsoll, 2022-09-08:
Actually I waas right the first time around :) Can't un-mark my comment as off topic so here is the reference again

See #246

amrsoll, 2022-09-15:
Now fixed with #246

closed 2022-09-15 by amrsoll