hlorus/CAD_Sketcher #280: [Bug]Nothing hovered

NoteToSelfFindGoodNickname, 2022-09-01:
### Contact Details

antoniocaroppos@gmail.com

### Description

Here is a video:

I am unable to change / edit the constraint.

I want to click it and enter a distance value, but it says "Nothing hovered".

https://www.youtube.com/watch?v=hjLXYwfy1Ec

### Addon Version

0.25

### Blender Version

3.25

### What platform are you running on?

Windows

NoteToSelfFindGoodNickname, 2022-09-01:
I could resolve it by clicking it in the list, but I guess the other way (clicking on it directly in the view) should work as well, so I guess it's really bug.

Here is how I do it by clicking in the list:

https://www.youtube.com/watch?v=8THICHqTZW4

hlorus, 2022-09-01:
Constraints are implemented as gizmos, therefore you can only interact with LMB. This is defined in blenders keymap, there's not much we can do frim the addon side here.

closed 2022-09-01 by hlorus