hlorus/CAD_Sketcher #287: [BUG]Not sure if a bug

NoteToSelfFindGoodNickname, 2022-09-02:
### Contact Details

antoniocaroppos@gmail.com

### Description

I don't know how to describe it, but it doesn't look right to me.
I have recorded a video here:

https://www.youtube.com/watch?v=Kx6rw57O65c

Sorry, it's 1:30 long, but I didn't know what would be important for you.

[intelligazethalter4.zip](https://github.com/hlorus/CAD_Sketcher/files/9474664/intelligazethalter4.zip)

### Addon Version

0.25

### Blender Version

3.25

### What platform are you running on?

Windows

amrsoll, 2022-09-02:
I believe this relates to #191

hlorus, 2022-09-02:
> I believe this relates to #191 

I would say so as well, let's close this as a duplicate then.

closed 2022-09-02 by hlorus