hlorus/CAD_Sketcher #290: [BUG] Python error when drawing 2D shape

vit21ik, 2022-09-03:
### Contact Details

vvlasuk@gmail.com

### Description

Description
Added 0.25 from zip and cannot create sketch due to python  errors.
Create sketch and go to sketch mode.
Select any 2D tool (arc, rectange) place cursor to surface and I get this:
[image]

Seems to die in icon_manager.py:73.

Addon Version
0.25.0

Blender Version
3.2.2

What platform are you running on?
Linux

### Addon Version

0.25

### Blender Version

3.2.2

### What platform are you running on?

Linux

hlorus, 2022-09-03:
> Seems to die in icon_manager.py:73.

Why exactly are you assuming this?

vit21ik, 2022-09-05:
>Seems to die in icon_manager.py:73.
Sorry, this line is not relevant to the bug. It seems like copy paste, my bad.

hlorus, 2022-09-05:
No problem. Could you explain what "place cursor to surface" means exactly? I assume it's a native blender operator, are you invoking it from a menu/ shortcut?
Also does it happen while in object mode?

vit21ik, 2022-09-05:
Steps are very simple: I selected 2d arc and clicked left mouse button anywhere in blender. As result - Report Error dialog as shown on attached screenshot.

hlorus, 2022-09-05:
I cannot reproduce this unfortunately. Maybe you could try to completely remove the addon and re-install...

amrsoll, 2022-09-15:
Let us know if you are able to reproduce it, preferably with a video showing the steps you took. You can reopen the issue :)

closed 2022-09-15 by amrsoll