hlorus/CAD_Sketcher #292: Installation problems on Mac

KeithSloan, 2022-09-05:
### Contact Details

keith@sloan-home.co.uk

### Description

Trying to follow the instructions at https://hlorus.github.io/CAD_Sketcher/installation/

Okay I click on the "Zip Archive" but being on a Mac this automatically gets expanded in $HOME/Downloads.
Okay I use compress to create a zip file and then I think I am able to install the Addon.

I assume it is not working because I do not have blender with solverspace installed.
( In the past I have installed solverspace as an app I assume from a dmg but it was some time ago )

The instructions say
It's recommended to use the addon with Blender installed from the official website.
What does that mean? I have to install a copy of Blender with solverspace already installed or just
that blender needs to have been installed from official website?

Then it gives the option Install from PyPi - Press. "Install from PyPI" but I see nothing like that to press.
Try manually installing with pip install py-slvs. Restart Blender but no solverspace in Bender Preferences.A bug happened!

### Addon Version

latest

### Blender Version

3.2.2

### What platform are you running on?

Mac OSX

hlorus, 2022-09-05:
I think it's necessary to manually re-zip it on mac in order to install blender addons.
You should be able to install and enable the addon regardless of the solver module. If the solver isn't present you'll be able to install it from the the addon preferences.

Also check out the [installation-help](https://discord.gg/r7g34TQW) channel on discord, there might be some mac people who can help

KeithSloan, 2022-09-05:
I already re-zipped and was able to install, was pointing out that instructions were not helpful in that respect.

solver is not present and cannot see how to install, not seeing in addon preferences, again instructions seemed very wanting in terms of an explanation and clarity. Like it says Press Install for PyPi but I see nothing like that.

Tried pip install py-slvs but that made no difference

hlorus, 2022-09-05:
Can you find the addon in the addons category of blender's preferences after the installation?

KeithSloan, 2022-09-05:
CAD-Solver No, solver No.

I seem to have a problem installing Addon via a zip file in Blender 3.2.2. Managed to install the FreeCAD addon by renaming the single py file as io_import_Freecad etc but not when a zip files.

hlorus, 2022-09-05:
This might help:
https://blenderartists.org/t/adding-addons-to-mac/605471

Otherwise you could try the installation process described under "Git" here: https://hlorus.github.io/CAD_Sketcher/installation/

KeithSloan, 2022-09-05:
"Otherwise you could try the installation process described under "Git" here: https://hlorus.github.io/CAD_Sketcher/installation/"

Those are the instructions I am trying to follow :-( :-(

hlorus, 2022-09-05:
> "Otherwise you could try the installation process described under "Git" here: https://hlorus.github.io/CAD_Sketcher/installation/"
> 
> Those are the instructions I am trying to follow :-( :-(

You won't have to zip it then. Simply clone the repository into your scripts/addons folder.

KeithSloan, 2022-09-05:
The CAD-SKetcher Addon is installed but does not work because of issue with solver see
([image]

hlorus, 2022-09-05:
Okay and what happens if you press the "Install from PIP" button and wait a minute?

KeithSloan, 2022-09-05:
Okay when restarted Blender it has appeared

hlorus, 2022-09-05:
Great 👍🏼

closed 2022-09-05 by hlorus

hlorus, 2022-09-05:
Thanks, I've applied your suggestions: #293