hlorus/CAD_Sketcher #299: Have multiple sketches for one mesh

theoryshaw, 2022-10-06:
**Motivation / Existing Problem**
This is perhaps more relative to [BlenderBIM](https://blenderbim.org/), but I have a feeling this functionality could be useful with a more agnostic use, as well. 

With BlenderBIM's implementation of IFC, a mesh can have multiple profiles associated with.  Example [here.](https://github.com/hlorus/CAD_Sketcher/issues/26#issuecomment-1109745905)

With the [current 'alpha' integration](https://github.com/hlorus/CAD_Sketcher/issues/26), one mesh can only have one CS sketch associated with it. 

**Description**
This feature request proposes developing an interface where 
An interface to be able to modify multiple profile per mesh object.

**Mockup**

[image]

hlorus, 2022-10-25:
I don't really get your proposal here, with boolean modifiers it is already possible to create a mesh from multiple sketches, doesn't that fit your needs?

Moult, 2022-10-27:
In IFC, a single object may be made out of multiple extrusions, each with its own profile. I'm not entirely sure how / if / what CAD Sketcher can do to make this type of integration more flexible, but let's perhaps keep this bug on hold until a workflow is more fleshed out. Potentially there is nothing to be done?

hlorus, 2022-10-27:
Sounds reasonable 👍🏼