hlorus/CAD_Sketcher #300: Specific sketch solidify result not correct [BUG]

flufy3d, 2022-10-10:
### Contact Details

flufy3d@gmail.com

### Description

A bug happened! if you do a sketch like this
[image]

the result  will be

[image]

it's good when I remove the solidify modifer
[image]

but look at the left side ,it's very strange.

pls have a look on this

Thanks

### Addon Version

0.25.0

### Blender Version

3.3.0

### What platform are you running on?

Windows

hlorus, 2023-01-23:
Closing as duplicate of #317

closed 2023-01-23 by hlorus

flufy3d, 2023-02-17:
it's solved with 0.27 version  thanks
[image]