hlorus/CAD_Sketcher #302: [BUG]

branlam, 2022-10-18:
### Contact Details

ichewhichew@gmail.com

### Description

When using the bevel on a point in CAD Sketcher, if you set a radius high enough where the second bevel may cut into the first bevel, it will cause it to crash blender.

[image]

The image above, I tried to set the bevel radius to 10 on the top and bottom points. Setting the top bevel to a radius of 10 after setting the bottom one to 10 would always crash my blender.

### Addon Version

0.25

### Blender Version

3.31

### What platform are you running on?

Windows

hlorus, 2022-10-19:
Thanks for the Report, i cannot reproduce this both on windows and linux. Does it also happen when there are no constraints?

branlam, 2022-10-19:
I tried it once again by simply making a small rectangle and setting the radius to 2. It crashed on me again from a fresh blender with only those steps. Is there other details I can provide that might help with explaining this issue for me?

hlorus, 2022-10-20:
Still cannot reproduce... a screen recording might be helpful. Also you might try to completely remove the addon and install the newest version.

branlam, 2022-10-20:
I was already on the newest version 0.25.0. But I did uninstall the addon and installed it again just to make sure.

Here is a screen capture of a fresh install of the addon and it causing blender to crash

https://user-images.githubusercontent.com/35716436/197034437-df67d355-5d39-4d7e-9ad3-e4f5f6487802.mp4

hlorus, 2022-10-20:
Sorry, i wasn't clear. This is likely resolved in the latest development version that you can get from GitHub directly. That version will soonish be pushed to gumroad along with some update notes.

branlam, 2022-10-20:
Ohh I see. I have downloaded the version from GitHub directly, the issue does not appear with that version 👍. Thank you very much

closed 2022-10-21 by hlorus