hlorus/CAD_Sketcher #303: [BUG] Hovering constrains breaks, restarting blender does not help.

clankill3r, 2022-10-20:
### Contact Details

clankill3r@gmail.com

### Description

There is a bug where hovering constrains does not work anymore. Restarting blender does not help. In the above file, if you hover over a radius of something or a distance, nothing will happen. If you remove all the constrains and then add a new one you can see that then the hovering over constrains works again.

There is a file in this discord post:

https://discord.com/channels/882152824249667594/966814731472093184/1032606978662486037

### Addon Version

0.25

### Blender Version

3.3

### What platform are you running on?

Windows

hlorus, 2022-10-20:
Thanks for the report, i've quickly looked at the file. The issue is that the "Constraints Selectability" is disabled (the property is displayed under debug settings if they're enabled in the addon preferences).
The property is used to allow selecting entities that are obscured by constraint gizmos, in v0.25 the left shift button is used to toggle the property (press sets it to False, release sets it to True).

You should be able to reset it by tapping the shift button. It shouldn't actually happen that this property persists to be False after the button is released, however the keymap for this is changed in the next version, this might help already.

clankill3r, 2022-10-20:
Ok thanks, I close the issue then.