hlorus/CAD_Sketcher #305: unable to select anything - Blender 3.3.1, Sketcher v0.25.0

Isaiah-Sugar, 2022-10-23:
Macos Catalina, 10.15.7
Blender 3.3.1, Cad Sketcher 0.25.0

In sketches, I can highlight things but not select them. 
I can still select them through the n-panel, but only one at a time? Could be misremembering but I thought I should be able to highlight multiple.

Ignore my slow laptop while trying to record the screen haha

https://user-images.githubusercontent.com/68121174/197417403-e1f4e49c-624c-4c9b-ac3c-a6dec15409cc.mp4
https://user-images.githubusercontent.com/68121174/197417784-dd27c1b1-1941-4722-b87a-d27e1b1064cb.mp4

hlorus, 2022-10-24:
Thanks for the report, the select tool got a mode property latley. This allows to either set/extend/subtract or toggle selection, similar to builtin selection tools. This is where the problems come from. 
I'll have to investigate why LMB doesn't select the entity underneath. As a workaround you can set the mode im the toolbar to toggle to get the old behavior.
Selection from the n panel should always toggle, this is just an oversight from me.

hlorus, 2022-10-24:
> Macos Catalina, 10.15.7 Blender 3.3.1, Cad Sketcher 0.25.0
> 
> .... I can still select them through the n-panel, but only one at a time? Could be misremembering but I thought I should be able to highlight multiple....

I've fix this one, could you try the new version and make sure to completely remove the old addon before installing the new one.

Isaiah-Sugar, 2022-10-25:
okay, thanks for changing the n-panel
I very appreciate your time / the work you've done!

closed 2023-01-17 by hlorus