hlorus/CAD_Sketcher #306: hole dimension constrain frozen[BUG]

adonishmedia, 2022-10-25:
### Contact Details

adonish.media@gmail.com

### Description

I created a hole in Cadsketcher. 
Next, I applied a dimensional constrain (alt o). The result was that the dimension cannot be moved or changed.
a workaround is to use the sketcher menu, look for the constrain (show/hide) then change dimemension. however, the dimension text stays inside the hole. I tried outside of a model but same results.
While following the CAD Modeling In Blender 3.2 | Using CAD Sketcher you tube, worked once. if I tried new holes then the issue arises.
I also seen intermittent issue with regular dimensions but more prominent with holes and arcs.

### Addon Version

0.25

### Blender Version

3.3.1

### What platform are you running on?

Windows

hlorus, 2022-10-25:
This might be the same issue as #303, could you check if that property is unticked as described in the comment of the mentioned task?

hlorus, 2023-01-17:
I'm assuming this has been resolved, feel free to reopen if the issue still persists.

closed 2023-01-17 by hlorus