hlorus/CAD_Sketcher #307: Constrain degrees of freedom

adonishmedia, 2022-10-25:
i noticed hard to determine degrees of freedom when contruction lines and actual lines are on top of each other.
Using sketcher menu, I kind of show and hide to try to find constrains and ofter delete and redo to clear the degrees of freedom flags.
Is there a simpler way.
Maybe add a color (red) to the checkmark in the constrain items to identify which constrain has a degree of freedom issue. once corrected we can see the checkmark turn to white or green.
Thanks.

hlorus, 2022-10-25:
> Maybe add a color (red) to the checkmark in the constrain items to identify which constrain has a degree of freedom issue. once corrected we can see the checkmark turn to white or green.
> Thanks.
> 

Unfortunately that's not possible as the solver doesn't support reporting which exact constraints have remaining degrees of freedom. Let's hope this will come at some point.

hlorus, 2026-08-20:
Reopening to track per-entity DOF highlighting. This needs the solver to report *which* entities are still free (currently it only gives the sketch-wide total); requested upstream in solvespace/solvespace#1756.