hlorus/CAD_Sketcher #310: [BUG]  Supplementry Angle Unit Input Doing strange Things

TheMakerTales, 2022-11-05:
### Description

When you try and control/input an angel to a supplementary angle things go a little weird. @hlorus hase taken a small look and said that it looked slightly "random" might be a good call for @amrsoll to take a look at it as you have the most experience working on this area of the code,

[image]

### Addon Version

latest

### Blender Version

3.3

### What platform are you running on?

Windows

TheMakerTales, 2023-01-15:
@amrsoll Any news on how this is going?

amrsoll, 2023-01-16:
> @amrsoll Any news on how this is going?

Ah I shouldn't have self assigned this as I was starting a new job and.. well  life happened ><

Anyway, I will try to dedicate some time tomorrow. If I don't get to it then here is a pointer:

It seems to be a purely UI problem stemming from a change I made where the angle property _should always_ refers to the direct angle, then the UI would convert that value if it wants to display the complementary angle.

I will unassign myself until I am sure that I have actually a good chance of finishing this.

amrsoll, 2023-01-16:
I still don't get how I missed this during tests, and I'm sorry this caused strain on the users and increased support

PR #322 (2023-01-24): Fix set_value for complementary angle constraint.
/!\ This might break some save files, as angles measured as "complementary" will be displayed as 180 - their.value 

This is a consequence of one of my previous problematic PRs, where the value of the constraints always refer to the same physical attribute. (e.g. the direct angle between 2 lines)

Although it introduces more functions, I still believe the idea is sound though, as we can guarantee that my_angle_property["value"] always refers to the direct angle.

Perhaps the functions `to/from_displayed_values()` do not fit in the class `Dimens [... cut]
commit 0b097948 (2023-01-24): Fix writes for complementary angle constraint values. (#322)

Closes #310
closed 2023-01-24 by hlorus