hlorus/CAD_Sketcher #317: [BUG] Solidify displacement issues

Sjefke03, 2023-01-10:
### Contact Details

stef.spooren@gmail.com

### Description

Ive been working on a project with everything working fine until today. Encountered the bug for the first time after trying to solidify a piece of my sketch. the top corners of my solidification seem to grow on the opposite side of my solidification direction.
[Example video](https://youtu.be/sVv3z1wXvkA)

### Addon Version

0.25

### Blender Version

3.3.1

### What platform are you running on?

Windows

flufy3d, 2023-02-17:
it's solved with 0.27 version  thanks
[image]

flufy3d, 2023-02-17:
> This seems to be an issue with the way blender calculates the normals of bezier curves. As there are sharp corners in the path it sometimes gets confused. I've fixed the issue once but unfortunately it's back again so i'm not sure to have this work while using bezier curves for convertion...
> 
> As a workaround i'd suggest to use geometry nodes to create a extrusion node. Maybe such a node could even be shipped with tha addon at some point.

with CAD_Sketcher version 0.27.0 and blender 3.4.1 the issue is sovled, love you work thank you!

flufy3d, 2023-02-17:
origin bug is here
#300

closed 2023-02-24 by hlorus