hlorus/CAD_Sketcher #318: [BUG] Delete dimensional constraint with drivers

Franckapik, 2023-01-11:
### Contact Details

fanch44@hotmail.com

### Description

When we delete dimensional constraint with drivers added, the values for all the list of dimensions are modified. 
There is a kind of shift .

[image]

### Addon Version

0.20

### Blender Version

3.3.1

### What platform are you running on?

Linux

Franckapik, 2023-03-03:
Hi @hlorus ! 
Thank you for having reported this issue in the good place ! 😉
It seems that we don't have any news? Nobody is working on this at this time, isn't it!?

hlorus, 2023-03-03:
Yeah it might take a while but i guess you can ask there about the progress...

ecosky, 2024-01-21:
Just encountered this bug myself lately when working on [some improvements](https://github.com/hlorus/CAD_Sketcher/pull/442) to driver related workflows.

Because that bug has been open for a while and there's no owner, project or milestone for it I think it's unlikely the Blender team will get to this problem soon. I would suggest working around it for now by manually destroying drivers associated with constraints before deleting them, but to my surprise there doesn't appear to be a Blender API that can do this. It seems the existing driver_remove function only removes drivers that match an input name, and there is no API to remove all drivers from what I could tell (see [blender source for driver_remove](https://projects.blender.org/blender/blender/src/commit/b92443e1d9c39ff869bf5662663e089fb7d13772/source/blender/python/intern/bpy_rna_anim.cc#L625)). There's also no way to enumerate the drivers to do get the information necessary to do individual driver_remove calls.

It might be worth submitting a patch to blender to provide a "driver_remove_all" function that would enable scripts to manually remove all drivers as part of their script's other functionality, while at the same time enabling people to work around this old problem in Blender.