hlorus/CAD_Sketcher #339: [BUG]

mjobson1, 2023-02-18:
### Contact Details

info@makeitexpress.com

### Description

There is a bug where you have an already fully defined sketch, but once you delete the main origin point (0,0) that references the x & y position of the drawing, it will still say the sketch is fully defined, but it will no longer be true. 

Even closing and opening the sketch will still say fully defined, but you are able to move the sketch in the x and y.

The simplest steps to repeat the bug :
1. Make a new sketch. leave default origin point (0,0) in place.
2. Make a square.
3. Make a diagonal line from top left of square to bottom right.
4. Add a midpoint to origin from that diagonal line. 
5. Make horizontal and vertical sides of square equal. 
6. Then give a dimension to one side of square. This should fully define the sketch.
7. Then delete the origin point. Either with delete key or right-click on point. (It won't pop-up with a warning. Unlike trying to delete the corner points of the square)
8. Select a corner point of the square and move freely. (notice it still says fully defined)
9. close and open sketch. Same results

See screenshots below:
[image]
[image]

Does it stop checking if the sketch is fully defined once it has happened the first time?

### Addon Version

0.27.0

### Blender Version

3.4.1

### What platform are you running on?

Windows

hlorus, 2023-02-24:
I assume the two screenshots are before &after deleting, right? The selected point doesn't seem to be removed for some reason then...

Also it looks like there's already a sketch, is it visible and co-planar to the Sketch.001? I could imagine that you're selecting &deleting the origin point of the other sketch as it's selectable. In that case we should make entities of the active sketch have a higher prio...

mjobson1, 2023-02-24:
Yes they are before and after deleting.

Yes in that example I did have another sketch, but the issues relates to the individual sketch, i didn't make the other sketch selectable.

I just repeated the bug by opening a brand new blender file, drawing a simple box and then a diagonal line, then making a midpoint relation, equaling horizontal and vertical lines of box and putting a dimension of 1.0m on box to fully define the sketch.

After that, you can delete that first centrepoint and it still says fully defined and you are able to move the box in x and y freely.

This is the bug. Very repeatable.

I can exit out the sketch and open it again and still says fully defined.

Just try to repeat the steps on your own. You'll see it clearly. Thanks.

hlorus, 2023-02-24:
Thanks for the report, should be fixed now.

commit 74f98135 (2023-02-24): Solve system after deleting entities

Fixes #339
closed 2023-02-24 by hlorus