hlorus/CAD_Sketcher #342: [BUG] Performance problems

rnovacek, 2023-02-26:
### Contact Details

rad.n@centrum.cz

### Description

I'm getting a terrible performance, every operation takes a couple of seconds (even selecting points, lines, etc). It was fine on this file initially but it was getting worse into a point it's not usable at all.

When I leave and hide the sketch, it's fine, no performance problems. 

The file is here: [box.zip](https://github.com/hlorus/CAD_Sketcher/files/10833381/box.zip)

### Addon Version

0.27.

### Blender Version

3.4.1

### What platform are you running on?

Linux

rnovacek, 2023-02-26:
Log from trying to open that file in Blender with CAD Sketcher debug on: [log.txt](https://github.com/hlorus/CAD_Sketcher/files/10833470/log.txt)

TheMakerTales, 2023-03-06:
Hmmm, I'm struggling to even open the .blend.
Do you happen to have a screen shot to show what it is that your making?
If it's something SUPPER complicated it might just be that were not at that point for hyper intricate designs

rnovacek, 2023-03-06:
Not too complicated [image]

I've started from scratch and it was perfectly fine, no slowdowns. So I think I've accidentally made some really hard to solve constraints in the first one. Maybe the solver would need some kind of timeout. Or maybe it runs too often (on every interaction, not just on changes). Or I'm completely wrong, sorry, I'm new to all this :)

lachlanA, 2023-05-25:
the slow down make's it unusable on Blender 3.5.1 and cad sketcher 0.27.1

Myrkskog, 2023-06-28:
> the slow down make's it unusable on Blender 3.5.1 and cad sketcher 0.27.1

I believe I was using the same version of Blender and CAD Sketcher. Experiencing performance issues. The slow behavior happens when first drawing in solvespace. Once the solvespace object is placed the editing/moving is performant.

Wondering if anyone else is experiencing similar. Running Ubuntu 22.10 testing both x11 and wayland. Same results for me.

BourneSuper, 2024-07-16:
it is slow when there are to many constrains. However, CPU and memory usage have not increased. Hide all constrains would operate faster.