hlorus/CAD_Sketcher #343: Copy Paste

speacock110, 2023-02-26:
### Contact Details

steve@stevepeacock-photography.com

### Description

Copy Paste does not work on my Mac (M1 all fully up to date)
Even though items are selected when I click on Cmd C I get a message no items selected

### Addon Version

0.27

### Blender Version

3.41

### What platform are you running on?

Mac OSX

hlorus, 2023-02-26:
What kind of message are you getting exactly?
Note that copy/pasting isn't supported in 3d, only when a sketch is active. Ctrl+c outside of a sketch will trigger blenders copy operator.

speacock110, 2023-02-26:
Marquee Select all of my Sketch then I press Cmd + C and i get the following screen

There are no entries in any of my clipboards content indicators and Cmd + V does nothing in either the same sketch or another clean one
￼
Steve

> 
> 
> What kind of message are you getting exactly?
> Note that copy/pasting isn't supported in 3d, only when a sketch is active. Ctrl+c outside of a sketch will trigger blenders copy operator.

-- 
Steve Peacock
+44(07932) 664679
http://www.stevepeacock-photography.com/
***@***.***
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Note : This E-Mail is sent in confidence for the addressee only.
Unauthorized recipients must preserve this confidentiality and
should please advise the sender immediately by telephone and
then delete the message without copying or storing it or disclosing
its contents to any other person.

We have taken all reasonable precautions to ensure that no viruses
are transmitted to any third party.

Any liability (in negligence or otherwise) arising from any party
acting, or refraining from acting on any information contained
in this e mail is hereby excluded.

Should you communicate with anyone at this address by e-mail,
you consent to us monitoring and reading any such correspondence.

Printing this email? Please think environmentally and only print when essential!

hlorus, 2023-02-26:
I'm not getting the pictures, could you try answering through the github website?

speacock110, 2023-02-26:
[image]

speacock110, 2023-02-26:
Posted

> 
> 
> I'm not getting the pictures, could you try answering through the github website?

-- 
Steve Peacock
+44(07932) 664679
http://www.stevepeacock-photography.com/
***@***.***
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Note : This E-Mail is sent in confidence for the addressee only.
Unauthorized recipients must preserve this confidentiality and
should please advise the sender immediately by telephone and
then delete the message without copying or storing it or disclosing
its contents to any other person.

We have taken all reasonable precautions to ensure that no viruses
are transmitted to any third party.

Any liability (in negligence or otherwise) arising from any party
acting, or refraining from acting on any information contained
in this e mail is hereby excluded.

Should you communicate with anyone at this address by e-mail,
you consent to us monitoring and reading any such correspondence.

Printing this email? Please think environmentally and only print when essential!

hlorus, 2023-02-26:
Did you intentionally close the issue?

speacock110, 2023-02-26:
no did not think i had just added the image 

sorry not a big GitHub user
Steve

> 
> 
> Did you intentionally close the issue?

-- 
Steve Peacock
+44(07932) 664679
http://www.stevepeacock-photography.com/
***@***.***
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Note : This E-Mail is sent in confidence for the addressee only.
Unauthorized recipients must preserve this confidentiality and
should please advise the sender immediately by telephone and
then delete the message without copying or storing it or disclosing
its contents to any other person.

We have taken all reasonable precautions to ensure that no viruses
are transmitted to any third party.

Any liability (in negligence or otherwise) arising from any party
acting, or refraining from acting on any information contained
in this e mail is hereby excluded.

Should you communicate with anyone at this address by e-mail,
you consent to us monitoring and reading any such correspondence.

Printing this email? Please think environmentally and only print when essential!

speacock110, 2023-02-26:
Reopened

hlorus, 2023-02-27:
The report "Copied 0 selected objects" is from blenders built-in copy operator. This is an indication that the addon operator "bpy.ops.view3d.slvs_copy" isn't triggered at all. 
In order to interact with addon entities the addons select tool has to be active. I'm guessing that might be the problem...

hlorus, 2023-03-01:
#348 might help if that was the problem as it makes some operators (like copy/paste) available in all addon tools and not just inside the selection tool.

closed 2023-06-01 by hlorus