hlorus/CAD_Sketcher #350: [BUG] Cannot Create Sketch

DevoidSloth, 2023-02-28:
### Description

When I click create sketch the grid pops up but does not allow me to select a plane for the sketch.

### Addon Version

latest

### Blender Version

2.93

### What platform are you running on?

Windows

hlorus, 2023-03-01:
You'll have to hover over the edge of the planes.

dariomangoni, 2023-03-06:
I don't know if we are having the same issue, but opening the System Console I'm getting a long list of the very same error

```
Python: Traceback (most recent call last):
  File "C:\Users\pc\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\CAD_Sketcher-main\gizmos\preselection.py", line 51, in test_select
    ensure_selection_texture(context)
  File "C:\Users\pc\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\CAD_Sketcher-main\draw_handler.py", line 43, in ensure_selection_texture
    draw_selection_buffer(context)
  File "C:\Users\pc\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\CAD_Sketcher-main\draw_handler.py", line 25, in draw_selection_buffer
    fb = gpu.state.active_framebuffer_get()
AttributeError: module 'gpu.state' has no attribute 'active_framebuffer_get'

location: <unknown location>:-1
```

@hlorus: even on the plane borders the same error is triggered, it does not matter where I click.

In Blender `Preferences>System>Cycles Render Devices` I selected CUDA and my graphic card is ticked in the list:
[image]

Is there anything to do?

I'm with Blender 2.93LTS, Win10 x64

hlorus, 2023-03-06:
Ahh i see, i'm getting the same error with version 2.93, this was caused by https://github.com/hlorus/CAD_Sketcher/commit/c78d0f5c60bb61a3ba94eab6d0a88cf3a0df9b47 which removed the usage of the bgl module which is going to be deprecated.
We'll have to either make it compatible with older versions or just bump the minimal required version to 3.0 which is IMO ok as we're not at 1.0 yet. Maybe @TheMakerTales can give his opinion here.

TheMakerTales, 2023-03-06:
Sure thing, I'll update the gumroad.
The newest LTS is 3.3 think so I think as long as we aim for that then it should be good

hlorus, 2023-03-06:
I've tested 3.0 and everything seems to be working, i'd go with this as long as there's no other reason.

timemaster5, 2023-07-11:
I also don't see workplanes on a Macbook Air M1, but this bugreport lead me to the GPU backend configuration, and since I've switched from Metal to Opengl, all was fine.

So at least on the Blender 3.6.0 on Mac M1 with Metal backend, I can say the workplanes are not visible.

egxn, 2023-12-04:
Also happens in 😟 
* Ubuntu 23.10
* CAD Sketcher version 0.27.3
* Blender 4.0.1

hlorus, 2024-07-25:
Yeah, thanks for pointing out!

closed 2024-07-25 by hlorus