hlorus/CAD_Sketcher #356: [BUG] Mesh creation fails when adding contraint

CobraA1, 2023-03-06:
### Contact Details

jeremiah.moss@gmail.com

### Description

This is how it looks before:

[image]

Added constraint to put lowest line on center point (midpoint constraint):

[image]

Completely ruins the mesh:

[image]

Blender file:
[Water bottle holder.zip](https://github.com/hlorus/CAD_Sketcher/files/10893996/Water.bottle.holder.zip)

### Addon Version

0.27.1

### Blender Version

3.4.1

### What platform are you running on?

Linux

TheMakerTales, 2023-03-06:
Hmm, personally I have a feeling this might be asking a bit much form mesh modelling.
I changed your value of the diameter to 65.499 mm and pretty much every works as I guess your looking for.
[image]

hlorus, 2023-06-01:
This isn't a bug of the addon but simply a limitation of blender's bezier curves which will fail with overlapping edges.

closed 2023-06-01 by hlorus