hlorus/CAD_Sketcher #363: [BUG] No module named 'py_slvs'

HrX03, 2023-03-12:
### Contact Details

dn.bianco03@gmail.com

### Description

Opening the console window i see a message stating `No module named 'py_slvs'` and of course the add-on page states the solver hasn't loaded properly.

When i try to install the module, either with PiP or with the wheel file i get the same log message on the console:
 ```
pip 23.0.1 from C:\Users\dnbia\AppData\Roaming\Python\Python310\site-packages\pip (python 3.10)
Requirement already satisfied: pip in c:\users\dnbia\appdata\roaming\python\python310\site-packages (23.0.1)
Processing c:\users\dnbia\downloads\py_slvs-1.0.5-cp310-cp310-win_amd64.whl
py-slvs is already installed with the same version as the provided wheel. Use --force-reinstall to force an installation of the wheel.
Name: py-slvs
Version: 1.0.5
Summary: Python binding of SOLVESPACE geometry constraint solver
Home-page: https://github.com/realthunder/slvs_py
Author: Zheng, Lei
Author-email: realthunder.dev@gmail.com
License: Gnu General Public License 3.0
Location: c:\users\dnbia\appdata\roaming\python\python310\site-packages
Requires:
Required-by:
Warning: Package should be available but cannot be found, check console for detailed info.
                Try restarting blender, otherwise get in contact.
```

It seems so the module is properly installed and should be loaded but, even if i were to restart blender, the module still doesn't get loaded with the same "no module named" error.

### Addon Version

latest

### Blender Version

3.4.1

### What platform are you running on?

Windows

soundpress, 2023-03-12:
Experiencing same issue with the same log text in system console
**Blender** 3.4.1, from blender.org, Running as Administrator
**CAD Sketcher** latest from https://hlorus.github.io/
**Platform** Win 10 x64
**py_slvs** is succesfully installable when installed with pip manually

hlorus, 2023-03-12:
You might wanna test this PR: https://github.com/hlorus/CAD_Sketcher/pull/249

This could solve your issue, would be great to get feedback.

soundpress, 2023-03-13:
Sadly it does not work. Resulted output is a bit different - seems pip succesfully installs py_slvs but failed with some its dependency:

```
No module named 'py_slvs'
pip 22.2.2 from C:\Program Files\Blender Foundation\Blender 3.4\3.4\python\lib\site-packages\pip (python 3.10)
Requirement already satisfied: pip in c:\program files\blender foundation\blender 3.4\3.4\python\lib\site-packages (22.2.2)
Collecting pip
  Using cached pip-23.0.1-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  WARNING: The scripts pip.exe, pip3.10.exe and pip3.exe are installed in 'C:\Users\Maxim\AppData\Roaming\Blender Foundation\Blender\3.4\scripts\addons\CAD_Sketcher-use-correct-pip\Python310\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed pip-23.0.1

[notice] A new release of pip available: 22.2.2 -> 23.0.1
[notice] To update, run: C:\Program Files\Blender Foundation\Blender 3.4\3.4\python\bin\python.exe -m pip install --upgrade pip
Collecting py-slvs
  Using cached py_slvs-1.0.5-cp310-cp310-win_amd64.whl (269 kB)
Installing collected packages: py-slvs
Successfully installed py-slvs-1.0.5
Name: py-slvs
Version: 1.0.5
Summary: Python binding of SOLVESPACE geometry constraint solver
Home-page: https://github.com/realthunder/slvs_py
Author: Zheng, Lei
Author-email: realthunder.dev@gmail.com
License: Gnu General Public License 3.0
Location: c:\users\maxim\appdata\roaming\blender foundation\blender\3.4\scripts\addons\cad_sketcher-use-correct-pip\python310\site-packages
Requires:
Required-by:
Warning: Package should be available but cannot be found, check console for detailed info. Try restarting blender, otherwise get in contact.
```

Enjoyed, 2023-03-15:
This issue #249  with it's corresponding PR will probably fix this issue. However, you can fix ***temporarily*** this issue by manually copying the py_slvs.
For me it was installed on:
`%appdata%\Python` (you can put this text directly on the windows explorer)
You should see a `py_slvs` folder in there. If not you might want to do a search for the folder name.
To ***temporarily*** make it work, copy this folder to `C:\Program Files\Blender Foundation\Blender 3.4\3.4\python\bin`

So the tree would be libe `...3.4\python\bin\py_slvs`. After that you can do the "PIP Install" button again and restart blender for it to be marked as registered

hlorus, 2023-03-17:
This should now be resolved with the 1.0.6 version of the py-slvs library, see: https://github.com/realthunder/slvs_py/issues/7

closed 2023-03-17 by hlorus