hlorus/CAD_Sketcher #366: [BUG] Package should be available but cannot be found

HarrisWilde, 2023-03-16:
### Contact Details

Harris20010411@outlook.com

### Description

A bug happened!
An error when registering solver module. I'd tried run Blender as administrator, but it still did't work.

Info:
```python
pip 23.0.1 from C:\Users\Harri\AppData\Roaming\Python\Python310\site-packages\pip (python 3.10)
Requirement already satisfied: pip in c:\users\harri\appdata\roaming\python\python310\site-packages (23.0.1)
Processing c:\users\harri\downloads\py_slvs-1.0.5-cp310-cp310-win_amd64.whl
py-slvs is already installed with the same version as the provided wheel. Use --force-reinstall to force an installation of the wheel.
Name: py-slvs
Version: 1.0.5
Summary: Python binding of SOLVESPACE geometry constraint solver
Home-page: https://github.com/realthunder/slvs_py
Author: Zheng, Lei
Author-email: realthunder.dev@gmail.com
License: Gnu General Public License 3.0
Location: c:\users\harri\appdata\roaming\python\python310\site-packages
Requires:
Required-by:
璀﹀憡: Package should be available but cannot be found, check console for detailed info.
                Try restarting blender, otherwise get in contact.
```

### Addon Version

0.27.1

### Blender Version

3.4.1

### What platform are you running on?

Windows

peter-ap, 2023-03-16:
I have the same issue.

hlorus, 2023-03-17:
This should now be resolved with the 1.0.6 version of the py-slvs library, see: https://github.com/realthunder/slvs_py/issues/7

closed 2023-03-17 by hlorus