hlorus/CAD_Sketcher #370: [BUG]: Can't pick workplane to create new sketch

BlazeCodeDev, 2023-03-21:
### Description

**What I expected:**

- Workplane highlights when hovering mouse over it and opens a new sketch on click.

**What actually happened:** 

- Workplane does not highlight on hovering and sketch creation gets cancelled no matter where I click.

**Steps to reproduce:**

1. Click "Add Sketch"
2. Move mouse over preview of workplanes
3. click on workplane

[image]

### Addon Version

0.27.1

### Blender Version

2.93.0

### What platform are you running on?

Windows

hlorus, 2023-03-22:
Workplanes are currently selectable on their edge only. This is the case because otherwise they would obscure each other as we don't have proper depth testing. I assume this is your problem, feel free to reopen the issue otherwise.

closed 2023-03-22 by hlorus

BlazeCodeDev, 2023-03-22:
Unfortunately this is not the solution for me. The workplanes also don't highlight on hover or get selected on click. Sadly I can't attach a screenshot, as for some reason the cursor is not being captured.

hlorus, 2023-03-22:
Ahh sorry, i just realized that you're on blender 2.93 while the minimal required version is now blender 3.0. You'll have to update the system to get it working.

BlazeCodeDev, 2023-03-22:
Ahh, alright thanks. Didn't see that.