hlorus/CAD_Sketcher #374: [BUG] Repeatedly pressing force update gives the sketch Redundant Constraints.

bEPHen, 2023-03-25:
### Contact Details

bephen@gmail.com

### Description

When I have a fully defined sketch and press force update 3 times I get Redundant Constraints which is hard to solve.
[image]
[image]

[Bug  forcing update three times.zip](https://github.com/hlorus/CAD_Sketcher/files/11069367/Bug.forcing.update.three.times.zip)

### Addon Version

0.27.1

### Blender Version

3.4.1

### What platform are you running on?

Linux

hlorus, 2023-03-30:
What happens when you remove the redundant constraints? Is it still fully defined?

hlorus, 2023-06-01:
This doesn't seem like a bug, removing one of the redundant constraints solves the error.

closed 2023-06-01 by hlorus