hlorus/CAD_Sketcher #377: [BUG]

jeremyvillalobos, 2023-04-11:
### Description

Blender 3.5.0
SketchCAD master e7bf7d228b0f34b773d934d2363e2a29b5f8d2f5
Ubuntu 20.04 ( stock Ubuntu, no kernel hacking or anything ) 
CPU: AMD FX(tm)-8150 Eight-Core Processor
GPU: Radeon R7 370

Stock Blender works Fine ( except for ray tracing which is just super slow )

When I start with the first use case for CAD Sketcher I do not see the square.

To reproduce:
1. Open blender.
2. Delete camera, light, cube
3. Press Ctrl+Shift+A
4. Select the X,Y plane ( Note, the YouTube video I saw will go into x,y plane view, but here I have to press number pad 1 to get it to look like on the YouTube vide, https://www.youtube.com/watch?v=1jNDLUDL0gc )
5. At this point I select the CAD Sketcher square 
6. When I go to draw there is no visual feedback that I am drawing a square.
Then, when I lift the mouse I only see pink small lines 

[image]

I can try by click and dragging 

[image]

I saw the square in black lines once upon a time.  Now looks broken. 

I can use some advice on how to erase the CAD Sketcher.

I have installed CADSketcher from source following the guide.
And also run 
```bash
blender --log-level -1  
```
But this does not give any leads to dig deeper.

Thanks

### Addon Version

e7bf7d228b0f34b773d934d2363e2a29b5f8d2f5

### Blender Version

3.5

### What platform are you running on?

Linux

jeremyvillalobos, 2023-04-12:
Here is the blender log after repeating the steps to reproduce the error:

```bash 
~$ blender --log-level -1 
Read prefs: /home/jeremy/.config/blender/3.5/config/userpref.blend
CAD_Sketcher:{INFO}: Logging into: /tmp/CAD_Sketcher-jeremy.log
Info: Deleted 1 object(s)

Info: Deleted 1 object(s)

Info: Deleted 1 object(s)

The context is  <class 'CAD_Sketcher.model.sketch.SlvsSketch'>
The context is  <class 'CAD_Sketcher.model.sketch.SlvsSketch'>

```

hlorus, 2023-04-12:
This might be a problem with the gpu where custom shaders are having an issue. Did you try the addon with previous blender versions?

jeremyvillalobos, 2023-04-13:
Tested with blender 3.0 and have the same issues. 

Then I tested 2.93.17, but for this version, I was not able to select one of the planes after pressing Ctrl+Shift+A

Let me know if there is a more specific version you want me to test. 

Do you know of a thread for the custom shader issue so I can take a look ?

jeremyvillalobos, 2023-04-13:
```location: <unknown location>:-1
Python: Traceback (most recent call last):
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/gizmos/preselection.py", line 51, in test_select
    ensure_selection_texture(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 43, in ensure_selection_texture
    draw_selection_buffer(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 25, in draw_selection_buffer
    fb = gpu.state.active_framebuffer_get()
AttributeError: module 'gpu.state' has no attribute 'active_framebuffer_get'

location: <unknown location>:-1
Python: Traceback (most recent call last):
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/gizmos/preselection.py", line 51, in test_select
    ensure_selection_texture(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 43, in ensure_selection_texture
    draw_selection_buffer(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 25, in draw_selection_buffer
    fb = gpu.state.active_framebuffer_get()
AttributeError: module 'gpu.state' has no attribute 'active_framebuffer_get'

location: <unknown location>:-1
Python: Traceback (most recent call last):
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/gizmos/preselection.py", line 51, in test_select
    ensure_selection_texture(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 43, in ensure_selection_texture
    draw_selection_buffer(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 25, in draw_selection_buffer
    fb = gpu.state.active_framebuffer_get()
AttributeError: module 'gpu.state' has no attribute 'active_framebuffer_get'

location: <unknown location>:-1
Python: Traceback (most recent call last):
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/gizmos/preselection.py", line 51, in test_select
    ensure_selection_texture(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 43, in ensure_selection_texture
    draw_selection_buffer(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 25, in draw_selection_buffer
    fb = gpu.state.active_framebuffer_get()
AttributeError: module 'gpu.state' has no attribute 'active_framebuffer_get'

location: <unknown location>:-1
Python: Traceback (most recent call last):
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/gizmos/preselection.py", line 51, in test_select
    ensure_selection_texture(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 43, in ensure_selection_texture
    draw_selection_buffer(context)
  File "/home/jeremy/blender_scripts/addons/CAD_Sketcher/draw_handler.py", line 25, in draw_selection_buffer
    fb = gpu.state.active_framebuffer_get()
AttributeError: module 'gpu.state' has no attribute 'active_framebuffer_get'

location: <unknown location>:-1
```
This is the log for the new error on 2.93.17.  I think I saw somebody else with this error so I will try to find and link.

hlorus, 2023-04-13:
Ahh the addon is supported for blender versions 3.0 and beyond so the error you got woth 2.93 is expected behavior. But good to know that it's failing with 3.0 as well.

hlorus, 2023-11-15:
There were some changes to shaders recently, could you try again with the newest addon version?