hlorus/CAD_Sketcher #378: [BUG] Precision on units for constraints too low (rounding happens)

alexfacciorusso, 2023-04-11:
### Contact Details

alexfacciorusso93@gmail.com

### Description

The precision assigned to the constraint's slider is too low. That causes constraints that should be shown precisely (e.g. 14.7mm) instead being rounded to the nearest unit (e.g. 15mm). When the slider is clicked, the text will show the full-precision value; see attached screenshots.

Even though the representation on the 3d view is correct (in that example, 14.70mm), I feel the precision should be increased in the panel's slider too.

[image]
[image]
[image]

### Addon Version

latest

### Blender Version

3.5.0

### What platform are you running on?

Windows

alexfacciorusso, 2023-04-11:
I could give this one a go and make a PR, if you all don't mind!

hlorus, 2023-04-11:
Thanks for the report. It's obviously great if you wanna take a look at this however be aware that this might not be easily possible as the precision is set when defining the property.

StrandedAlien, 2023-04-21:
Brave man, Alex, i wish you success. The low precision has been bugging CAD interested Blender users for a long time. I suggest you ask at the Blender dev forum https://devtalk.blender.org/ before you even start. My guess is it belongs to "Core Module". I had trouble with an imported 3D Maps data offset when i wanted to join the pieces. Found out it was due to Blenders low precision.

closed 2023-06-01 by hlorus