hlorus/CAD_Sketcher #381: [BUG] Force Update causes modifiers to be lost

MarkSwanson, 2023-04-17:
### Contact Details

mark.swanson.sw@gmail.com

### Description

I'm going through the nice tutorial and if you click the 'Force Update' button at t=1101
https://youtu.be/941iYxf3HuM?t=1101

Then all object modifiers are lost for all sketch objects.
Fyi clicking 'Force Update' has caused the object modifiers to get lost at other times.

### Addon Version

0.27.1

### Blender Version

3.5.0

### What platform are you running on?

Linux

hlorus, 2023-04-18:
Hey there, i cannot reproduce this by simply clicking "Force Update". The modifiers are lost when changing the convert type but that's expected behavior as the different geometry types don't support the same modifiers.

hlorus, 2023-04-18:
The problem here seems to be the bool tool. The auto boolean seems to directly manipulate the mesh rather than adding a modifier. With the brush boolean it seems to work as expected as long as the brush isn't applied.

MarkSwanson, 2023-04-18:
Ah, thank you for explaining. The brush boolean indeed does seem to work perfectly for me.
Thank you for creating CAD Sketcher. It seems amazing so far!

closed 2023-04-18 by MarkSwanson