hlorus/CAD_Sketcher #382: [BUG]Error on install v0.27.1

jikometrix, 2023-05-02:
### Contact Details

mbrando@jikometrix.net

### Description

I tried to install v0.27.1 cad sketcher on Blender 3.5.1 on Mac M1 with os 13.3.1
I get the attached error. 
[image]
[image]

### Addon Version

0.27.1

### Blender Version

3.5.1

### What platform are you running on?

Mac OSX

Benjemen69, 2023-05-03:
Same issue here. Tried downloading and installing, the issue still persists.

hlorus, 2023-05-03:
Closing as duplicate of #375

closed 2023-05-03 by hlorus