hlorus/CAD_Sketcher #389: Bevel Tool Supporting Lines For Constrained Points

TheMakerTales, 2023-06-08:
Currently, when adding a bevel to a point that already has a dimensional constant it's behavior is this:

[image]

Let's add some supporting construction like creation for point that are beveled that are already constrained in some way like this:

[image]

I'm unsure if this should be "automatic" behavior, toggleable or modifier key initiated i.e. pressing shift while using the bevel tool

TheMakerTales, 2023-06-22:
Just to be clear the resulting construction lines should be parallel to the original line thus supporting lines in any orientation.

TimStrauven, 2024-04-18:
This behavior is linked to:
 Cannot set Distance constraint to horizontal or vertical for lines [BUG] #459 
It should be half solved if the PR on it is accepted:
 Checks for duplicate constraints. #461 

The reason it happened is also because the distance constraint is set onto the line instead of its endpoints.
New behavior after merge of that PR would look like this:
[image]

So if the bevel tool would then add a coincident from the point to each line it should be solved as well I think.

hlorus, 2024-04-19:
Good catch, can you open a PR?

TimStrauven, 2024-04-19:
If you also want the lines to be added, this seems to work well  and I think it fully covers the initial request:

```Python
        # Remove original point if not referenced
        if not is_entity_referenced(point, context):
            context.scene.sketcher.entities.remove(point.slvs_index)
        else:
            ssc.add_coincident(point, seg1, sketch)
            ssc.add_coincident(point, seg2, sketch)

            # add reference construction lines
            sse = context.scene.sketcher.entities
            for i in range(0, 2):
                if isinstance(self.connected[i], SlvsLine2D):
                    target = sse.add_line_2d(point, self.points[i], sketch)
                    target.construction = True
                elif isinstance(self.connected[i], SlvsArc):
                    target = sse.add_arc(sketch.wp.nm, self.connected[i].ct, self.points[i], point, sketch)
                    target.construction = True
                    if target.angle > HALF_TURN:
                        target.invert_direction = True
´´´

![image](https://github.com/hlorus/CAD_Sketcher/assets/96440176/d35c1903-4907-43e5-929c-b49f731ed037)
![image](https://github.com/hlorus/CAD_Sketcher/assets/96440176/93c4e72f-478e-40da-9846-614f74655c85)
![image](https://github.com/hlorus/CAD_Sketcher/assets/96440176/e8409cee-b065-4194-a2cc-d0c8f942f805)

Would you prefer it this way or rather without the construction lines?

closed 2024-12-19 by hlorus