hlorus/CAD_Sketcher #391: Documentation lacks important step

noxer, 2023-06-13:
In the "Getting Started" section of the website [1], in the "Create a Scetch" step it says "In the 3d Viewport open the sidebar and change to the "Sketcher" tab.". The screenshot suggests there is a bar on the right side of the screen. Unfortunately it doesn't say *how* to open the sidebar, making it impossible for beginners to follow.

[1] https://hlorus.github.io/CAD_Sketcher/getting_started/

yokonzo, 2023-07-23:
Can confirm, took me 30 minutes to find out you meant the little arrow symbol next to the 3d veiwport controls, dragging that out pulled out the sidebar and allowed me to see the sketcher tab

hlorus, 2023-11-13:
Closing 67c2dbf4478cf1b482a44a1dcca4e4a9cb9c3eac

closed 2023-11-13 by hlorus