hlorus/CAD_Sketcher #395: [BUG] Crash on creating sketch

ffernn-dev, 2023-06-17:
### Contact Details

hihi.fernguy@gmail.com

### Description

Just installed on Arch linux, installed the pip module with `paru -S python-py-slvs` (rather than using pip as advised by Arch), and am getting a crash while attempting to create a sketch in the sidebar.
```
/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slvs_Constraint; _Alloc = std::allocator<Slvs_Constraint>; reference = Slvs_Constraint&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    105499 IOT instruction (core dumped)  blender
``` I get this error in my console when it crashes

### Addon Version

0.27.2 - latest

### Blender Version

3.5.1

### What platform are you running on?

Linux

hlorus, 2023-06-18:
Might be the wrong solver, this is the correct one:
https://pypi.org/project/py-slvs/

JulianGmp, 2023-06-27:
I can confirm that this issue occurs for me as well.
I noticed that this also happens when I press the "Force Update" button, with similar output:
[image]
```
> blender --log-level -1
Read prefs: "/home/julian/.config/blender/3.6/config/userpref.blend"
Failed to open dir (No such file or directory): /run/user/1000/gvfs/
CAD_Sketcher-main:{INFO}: Logging into: /tmp/CAD_Sketcher-main-julian.log
/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slvs_Param; _Alloc = std::allocator<Slvs_Param>; reference = Slvs_Param&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
zsh: IOT instruction (core dumped)  blender --log-level -1
```

I am using the [python-py-slvs AUR package](https://aur.archlinux.org/packages/python-py-slvs), which uses the same source as the pypi package.

### Addon Version
0.27.2 (latest gumroad download)

### Blender Version
3.6.0

### What platform are you running on?
Arch Linux 2023-06-27, all packages updated.

JulianGmp, 2023-06-27:
Update: the issue is resolved when using py-slvs 1.0.6.

I don't know how easy/hard it is to do in python, but I would suggest adding a version check to the addon to see if it's running with the right version of the module.

bury52, 2023-07-03:
```

Read prefs: "/home/bogusz/.config/blender/3.6/config/userpref.blend"
CAD_Sketcher-main:{INFO}: Logging into: /tmp/CAD_Sketcher-main-bogusz.log
Info: Deleted 3 object(s)

/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slvs_Constraint; _Alloc = std::allocator<Slvs_Constraint>; reference = Slvs_Constraint&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
fish: Job 1, 'blender' terminated by signal SIGABRT (Przerwij)

```

```bash
yay -S python-py-slvs-git
```

AUR - [python-py-slvs-git-1.0.6.r1.gb704719-1](https://aur.archlinux.org/packages/python-py-slvs-git)

Addon Version
0.27.2 (latest gumroad download)

Blender Version
3.6.0

What platform are you running on?
Arch Linux 2023-07-03, all packages updated.

ffernn-dev, 2023-07-05:
Hey sorry I haven't responded to this. I was able to get it working by forcefully installing the pypi package, but with complain from Arch. It is much preferable to use AUR or official Arch packages for python software due to the way the system operates, which is why pressing the "install slvs" button in the settings menu doesn't do anything (it errors when trying to run pip install)

chuck-flowers, 2023-07-07:
I'm seeing the same issue on Arch using both the python-py-slvs-git and the python-py-slvs AUR packages (the latter which I maintain so if there's an issue with the package let me know). When I run blender from the command line, I see the following error message which might provide some insight.

```
Read prefs: "/home/cflowers/.config/blender/3.6/config/userpref.blend"
cad-sketcher:{INFO}: Logging into: /tmp/cad-sketcher-cflowers.log
/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slvs_Constraint; _Alloc = std::allocator<Slvs_Constraint>; reference = Slvs_Constraint&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```

linkdom, 2023-07-09:
I can confirm that I get the exact same error with both AUR packages, if I can provide any further help let me know and I will do my best!

Blender 3.6.0
        build date: 2023-07-02
        build time: 18:04:48
        build commit date: 2023-06-27
        build commit time: 08:08
        build hash: c7fc78b81ecb
        build platform: Linux
        build type: Release

CAD Sketcher 0.27.2

hlorus, 2023-07-10:
> I'm seeing the same issue on Arch using both the python-py-slvs-git and the python-py-slvs AUR packages (the latter which I maintain so if there's an issue with the package let me know). When I run [...]

I'm guessing the package doesn't include the latest fix, see: https://github.com/realthunder/slvs_py/commit/3df1c875d7be9fa26f753a654b5305751ebeafa6

Jakobu5, 2023-07-29:
Is there any update regarding the crashing? Blender crashes with version py-slvs 1.0.6 (installed via AUR):

```bash
blender --log-level 1
Read prefs: "/home/jakobu5/.config/blender/3.6/config/userpref.blend"
Failed to open dir (No such file or directory): /run/user/1000/gvfs/
Traceback (most recent call last):
  File "/usr/share/blender/3.6/scripts/modules/addon_utils.py", line 333, in enable
    mod = import(module_name)
          ^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'cad-sketcher'
CAD_Sketcher-main:{INFO}: Logging into: /tmp/CAD_Sketcher-main-jakobu5.log
/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slvs_Constraint; _Alloc = std::allocator<Slvs_Constraint>; reference = Slvs_Constraint&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)

[jakobu5@archlinux ~]$ blender --version
Blender 3.6.0
        build date: 2023-07-07
        build time: 13:20:07
        build commit date: 2023-06-27
        build commit time: 08:08
        build hash: c7fc78b81ecb
        build platform: Linux
        build type: Release
        [...]

[jakobu5@archlinux ~]$ pip list
Package      Version
------------ -------
[...]
py-slvs      1.0.6
[...]
```

chuck-flowers, 2023-08-01:
@Jakobu5 I don't think we've gotten an answer on what the problem is yet. Your error seems to imply that `cad-sketcher` is not installed given the message `ModuleNotFoundError: No module named 'cad-sketcher'`

It might be worth reinstalling the `cad-sketcher` plugin to see if the error message changes.

Aggeloz, 2023-08-01:
I have the exact same but with this terminal output as well 

`/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slvs_Constraint; _Alloc = std::allocator<Slvs_Constraint>; reference = Slvs_Constraint&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    12339 IOT instruction (core dumped)  blender`

Jakobu5, 2023-08-02:
> @Jakobu5 I don't think we've gotten an answer on what the problem is yet. Your error seems to imply that `cad-sketcher` is not installed given the message `ModuleNotFoundError: No module named [...]

I think that’s because i’ve enabled the plugin right before clicking on „New Sketch“ in the CAD Sketcher UI.
After enabling the plugin and reopening Blender it crashes exactly the same, without the `ModuleNotFoundError`:

```bash
blender --log-level 1
Read prefs: "/home/jakobu5/.config/blender/3.6/config/userpref.blend"
CAD_Sketcher-main:{INFO}: Logging into: /tmp/CAD_Sketcher-main-jakobu5.log
/usr/include/c++/13.1.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slvs_Constraint; _Alloc = std::allocator<Slvs_Constraint>; reference = Slvs_Constraint&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```

The Logfile `CAD_Sketcher-main-jakobu5.log` is empty.

vitrvvivs, 2023-08-19:
Easiest way to reproduce is with this
```python
>>> import py_slvs.slvs
>>> py_slvs.slvs.System().solve()
```

@chuck-flowers
Dirty Solution:
```diff
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -40,6 +40,7 @@ prepare() {
 build() {
        cd "$srcdir/$pkgname"

+       export CXXFLAGS="-U_GLIBCXX_ASSERTIONS"
        python setup.py build
 }
```

Reason:
The bug is upstream in solvespace.
In the Arch default /etc/makepkg.conf : `CXXFLAGS="$CFLAGS -Wp,-D_GLIBCXX_ASSERTIONS"`.
`-D_GLIBCXX_ASSERTIONS` adds out-of-bounds checks when accessing vectors.  This is generally a good thing, better than undefined behaviour.

I believe the offending line is in the python-C++ interface. https://github.com/realthunder/solvespace/blob/526a260b0c45586c0319de208fd7e97c43c49bf4/src/swig/slvs_swig.hpp#L51 . It's the only place in the project with an std::vector of Constraints. When there are no constraints yet added, that [0] access should fail.

Vermoot, 2023-08-20:
Still having this issue as well. Downloaded the latest CAD sketcher (0.27.2), and different versions of `py-slvs` from the AUR, nothing fixed it.

chuck-flowers, 2023-08-20:
@vitrvvivs Thanks for sharing this. I wonder if the default values `CXXFLAGS` for Arch Linux's `/etc/makepkg.conf` have changed recently. I'll look into this if I have time later today. If I find a quick solution, I'll post here. Finding a more long term solution for everyone using the AUR directly might take a little bit more time.

chuck-flowers, 2023-08-23:
@vitrvvivs sorry for the delay in getting back to this. Your solution with the `CXX_FLAGS` seems to work. I modified it slightly to include the `-Wp` and `$CFLAGS` values originally defined in `/etc/makepkg.conf`. I have published a change to the `python-py-slvs` package. I don't control the `python-py-slvs-git` package so I can't update that one.

If anyone else has any problems they believe could be related to my package, let me know.

Qwyntex, 2025-05-01:
Cmake has removed compatibility with cmake < 3.5, which causes a build failure in the `python-py-slvs` package.

[image]

`python-py-slvs-git` builds but the Blender still crashes when creating a new sketch, given that there has been no solution implemented in that package.

I was able to build using the `-DCMAKE_POLICY_VERSION_MINIMUM=3.5` flag

Qwyntex, 2025-05-01:
I can also verify the addon working in blender with my manually compiled package.
It'd be great if you could edit the package :)
Thank you for your work on this plugin!