hlorus/CAD_Sketcher #401: [BUG] Constraints are not visible in viewport UI nor in 'Constraints' panel

zanovis, 2023-07-08:
### Description

Constraints are not visible in viewport/side panel. Additionally, attempting to delete the sketch throws this error:

[image]

```
Python: Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\ui\panels\constraints_list.py", line 98, in draw
    draw_constraint_listitem(context, col, c)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\ui\panels\constraints_list.py", line 15, in draw_constraint_listitem
    index = context.scene.sketcher.constraints.get_index(constraint)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\model\group_constraints.py", line 120, in get_index
    list = getattr(self, constr.type.lower())
AttributeError: 'SlvsConstraints' object has no attribute ''
```

### Addon Version

.27

### Blender Version

3.6

### What platform are you running on?

Windows

hlorus, 2023-07-09:
Are you able to reproduce the error and could submit the steps to reproduce?

zanovis, 2023-07-09:
Yes, it is reproducible. I haven't been able to get it to show constraints
since installing (steps following the installation steps indicated).

hlorus, 2023-07-09:
I'll need the full content of the terminal to help, see: https://hlorus.github.io/CAD_Sketcher/advanced/#console-output

zanovis, 2023-07-09:
Does this help?
```
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\ui\panels\constraints_list.py", line 110, in draw
    draw_constraint_listitem(context, col, c)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\ui\panels\constraints_list.py", line 15, in draw_constraint_listitem
    index = context.scene.sketcher.constraints.get_index(constraint)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\model\group_constraints.py", line 120, in get_index
    list = getattr(self, constr.type.lower())
AttributeError: 'SlvsConstraints' object has no attribute ''
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 159, in draw
    constraint = self._get_constraint(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
  File "C:\Program Files\Blender Foundation\Blender 3.6\3.6\scripts\modules\bpy_types.py", line 741, in __getattribute__    return super().__getattribute__(attr)
AttributeError: 'VIEW3D_GT_slvs_constraint' object has no attribute 'index'
```

hlorus, 2023-07-10:
> Does this help?

Not really, this is just the draw function that complains about a problem but i can't see what caused the problem in the first place. Usually there's a different traceback at the top of the console...

zanovis, 2023-07-22:
How's this?

`StructRNA of type BlenderKitDisclaimerOperator has been removed
StructRNA of type BlenderKitDisclaimerOperator has been removed
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 114, in refresh
    self.setup(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 73, in setup
    gz.index = context.scene.sketcher.constraints.get_index(c)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\model\group_constraints.py", line 120, in get_index
    list = getattr(self, constr.type.lower())
AttributeError: 'SlvsConstraints' object has no attribute ''
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 159, in draw
    constraint = self._get_constraint(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
  File "C:\Program Files\Blender Foundation\Blender 3.6\3.6\scripts\modules\bpy_types.py", line 741, in __getattribute__    return super().__getattribute__(attr)
AttributeError: 'VIEW3D_GT_slvs_constraint' object has no attribute 'index'
StructRNA of type BlenderKitDisclaimerOperator has been removed
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 159, in draw
    constraint = self._get_constraint(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
  File "C:\Program Files\Blender Foundation\Blender 3.6\3.6\scripts\modules\bpy_types.py", line 741, in __getattribute__    return super().__getattribute__(attr)
AttributeError: 'VIEW3D_GT_slvs_constraint' object has no attribute 'index'
StructRNA of type BlenderKitDisclaimerOperator has been removed
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 159, in draw
    constraint = self._get_constraint(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
  File "C:\Program Files\Blender Foundation\Blender 3.6\3.6\scripts\modules\bpy_types.py", line 741, in __getattribute__    return super().__getattribute__(attr)
AttributeError: 'VIEW3D_GT_slvs_constraint' object has no attribute 'index'
StructRNA of type BlenderKitDisclaimerOperator has been removed
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 159, in draw
    constraint = self._get_constraint(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
  File "C:\Program Files\Blender Foundation\Blender 3.6\3.6\scripts\modules\bpy_types.py", line 741, in __getattribute__    return super().__getattribute__(attr)
AttributeError: 'VIEW3D_GT_slvs_constraint' object has no attribute 'index'
StructRNA of type BlenderKitDisclaimerOperator has been removed
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 159, in draw
    constraint = self._get_constraint(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
  File "C:\Program Files\Blender Foundation\Blender 3.6\3.6\scripts\modules\bpy_types.py", line 741, in __getattribute__    return super().__getattribute__(attr)
AttributeError: 'VIEW3D_GT_slvs_constraint' object has no attribute 'index'
StructRNA of type BlenderKitDisclaimerOperator has been removed
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 159, in draw
    constraint = self._get_constraint(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
  File "C:\Program Files\Blender Foundation\Blender 3.6\3.6\scripts\modules\bpy_types.py", line 741, in __getattribute__    return super().__getattribute__(attr)
AttributeError: 'VIEW3D_GT_slvs_constraint' object has no attribute 'index'
StructRNA of type BlenderKitDisclaimerOperator has been removed
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\constraint.py", line 159, in draw
    constraint = self._get_constraint(context)
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
[... cut ...]
  File "C:\Users\Noah\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\CAD_Sketcher-main\gizmos\base.py", line 10, in _get_constraint
    self.type, self.index
  File "C:\Program Files\Blender Foundation\Blender 3.6\3.6\scripts\modules\bpy_types.py", line 741, in __getattribute__    return super().__getattribute__(attr)
AttributeError: 'VIEW3D_GT_slvs_constraint' object has no attribute 'index'
StructRNA of type BlenderKitDisclaimerOperator has been removed
StructRNA of type BlenderKitDisclaimerOperator has been removed`

hlorus, 2023-07-23:
Looks like there are some issues with the blenderkit addon. Have you already tried if the issue persists if you disable that?