hlorus/CAD_Sketcher #404: [BUG] Can't select Arcs

Nexion22, 2023-07-13:
### Description

Hello,
I'm having trouble selecting the arc I've just created. To select it I have to create a new one and over the mouse on it. The previous arc will be selected. Very strange, isn't it?

I've already tried to reset Blender
py_slvs version is 1.0.6

[image]

### Addon Version

0.27.2

### Blender Version

3.6.0

### What platform are you running on?

Linux

ClayShoaf, 2023-07-15:
I just tried this using the same versions as you in Ubuntu and I'm not having this problem.
[Screencast from 07-14-2023 10:55:39 PM.webm](https://github.com/hlorus/CAD_Sketcher/assets/31578812/4cddde49-780c-4947-b7ef-ca4cd8a1bfb3)

I can not get Ubuntu's screen record to show the mouse while I'm doing it, but the correct arcs are being selected.

Nexion22, 2023-07-15:
I think it might be hardware-dependent, because I have 3 computers with the same ArchLinux installation, same software versions, but on my two Thinkpads X220 and X230T I have the problem. And on my Thinkpad W541 at work I don't.

ClayShoaf, 2023-07-15:
I know this is a little off topic, but how in the world are you running blender in a usable way on a X220?

Nexion22, 2023-07-15:
I mainly use blender for modeling 3D printing parts. For parts that aren't too complex, it works. :)
But it's not ideal, I confess.